lvgl/lvgl #1603: Text Align does not work properly when TextArea.text is empty

weixiongmei, 2020-06-22:
When the TextArea.text is empty, the cursor does not align properly, it always aligning to the left. LVGL7.0.2

kisvegabor, 2020-06-22:
Please fill the template to provide more information. A screenshot would be really helpful in this case.
```
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

A clear and concise description of what the bug is.

**To Reproduce**

Please provide a small, independent code sample that can be used to reproduce the issue. Ideally this should work in the PC simulator unless the problem is specific to one platform.

**Expected behavior**

A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

Add any other context about the problem here.

```

weixiongmei, 2020-06-23:
Information have been updated, please help~~ Thanks

kisvegabor, 2020-06-24:
Thanks for the update.
I fixed it here: 63aba1b3

kisvegabor, 2020-07-08:
As the issue seems to be solved I close it.

If it's still not working please reopen the issue and comment here.

closed 2020-07-08 by kisvegabor