lvgl/lvgl #1604: Circular tab view

kisvegabor, 2020-06-22:
Continue the [related Feature request topic](https://forum.lvgl.io/t/tab-view-circular-scrolling/2473) from the Forum.

The task is to make the scrolling of tabview circular, i.e. swiping right after the last tab shows the first tab. I'm not sure what is the best was to implement it:
1. Dynamically arrange the tab pages to always have an adjacent page.
2. Introduce some magic into LVGL to transform coordinates in a circular way. 

Ideas are welcome :slightly_smiling_face:

embeddedt, 2020-06-22:
> Dynamically arrange the tab pages to always have an adjacent page.

This seems like the simpler approach to me. I'm looking into how this would be implemented. From what I can tell, the order of the tabs is decided by a page, so I would need to reorder the children of that page?

Then, I might need to modify the logic in `tabview_scrl_signal` to take into account that the tabs aren't arranged in order anymore. Is that correct?

kisvegabor, 2020-06-23:
> This seems like the simpler approach to me. 

For me too.

> I can tell, the order of the tabs is decided by a page, so I would need to reorder the children of that page?

The pages are aligned into a row with a layout and order is determined by the order of creation. It's not strictly necessary to use a layout because `tabview_realign` is always called if needed and the positioning can happen there manually too.  

> Then, I might need to modify the logic in tabview_scrl_signal to take into account that the tabs aren't arranged in order anymore. Is that correct?

I think the goal is to ensure that always there is page next to the first/last page. At first look, it seems `lv_tabview_set_tab_act` is called every time a new page is loaded either by sliding or clicking the header. Hopefully, the logic can be added only there.

closed 2020-07-22 by stale[bot]