lvgl/lvgl #1605: Style performance issue

gangheivt, 2020-06-23:
To redraw an image, the style of images need to be accessed. It would search styles in parents as well if the style is not set. I have tried to refresh one image, it called lv_obj_get_style_xxx more than 20 times. This would have 400-500us extra delay in our 48MHz MCU. Is there a way to improve the performance?

embeddedt, 2020-06-23:
Increasing your display buffer size will reduce the number of times styles need to be accessed for a single object.

gangheivt, 2020-06-23:
In our implement, we need to rotate the image frequently.  The image need to be refreshed. lv_img_design is called every time, and it has a lot of style access. If image style is empty, it will go through its parents. This caused a lot of delay.

kisvegabor, 2020-06-24:
I've played a lot with caching styles properties but always concluded that they don't have notable effect on performance. Although it would be logical to have a measurable effect.  (I used more powerful MCUs for testing).

How could you measure that 400-500 us? 
It means 400/20 = 20 us for every `lv_obj_get_style_xxx`. On 48 MHz it's ~ 1000 instructions. Which seems possible. 

Anyway, I'm very open to optimize the style's performance.

gangheivt, 2020-06-24:
In lv_img_design, I counted cycles before reaching lv_draw_img. The UI is a simple image that try to rotate every 30ms. Since our code is  running in flash(both code and image), I suspect there is a cache miss that might make things worse.  I hope we could have a cache friendly design in style, such as isolating those functions so I could put in ROM/RAM with minimal cost.

embeddedt, 2020-06-24:
> I've played a lot with caching styles properties but always concluded that they don't have notable effect on performance. 

@kisvegabor Have you tested it with smaller buffers (i.e. 1/10th or 1/20th of the display's size)? These sizes are where 7.0 is significantly slower than 6.1, because the styles get recalculated for every chunk of the display that is redrawn.

gangheivt, 2020-06-25:
@embeddedt , I am using full screen buffer in this case.

kisvegabor, 2020-06-26:
@gangheivt 
Please try these settings in `lv_conf.h`:
- `LV_USE_BLEND_MODES 0`: skips some style queries 
- `LV_USE_OPA_SCALE 0`: also skip some style queries
- `LV_ATTRIBUTE_FAST_MEM` to place performance-critical function into RAM

@embeddedt 
As I remember I haven't tested it with really small buffers (such as 1/20). 
Meanwhile, I was thinking about another method to speed up style accesses. Now the properties are stored in arbitrary order (in order as properties were added). By storing them in order for some special search algorithms could be used to skip most of the checks.

embeddedt, 2020-06-26:
I re-ran `gprof` on the PC simulator to see what the slowest code paths are there. It appears that `lv_color_fill` and the static `get_property_index` function are the slowest functions at the moment.

```
  %   cumulative   self              self     total           
 time   seconds   seconds    calls  ms/call  ms/call  name    
 48.39      0.15     0.15   145840     0.00     0.00  lv_color_fill
  9.68      0.18     0.03   279097     0.00     0.00  get_property_index
  6.45      0.20     0.02     5540     0.00     0.01  draw_letter_normal
  3.23      0.21     0.01   502749     0.00     0.00  ent_alloc
  3.23      0.22     0.01   103746     0.00     0.00  _lv_memset_00
  3.23      0.23     0.01    65705     0.00     0.00  _lv_blend_fill
  3.23      0.24     0.01    61395     0.00     0.00  lv_draw_mask_line
  3.23      0.25     0.01    60136     0.00     0.00  fill_normal
  3.23      0.26     0.01    40772     0.00     0.00  _lv_memcpy_small
  3.23      0.27     0.01    19951     0.00     0.00  _lv_style_list_get_ptr
  3.23      0.28     0.01    12322     0.00     0.00  lv_bidi_letter_is_neutral
  3.23      0.29     0.01    11515     0.00     0.00  _lv_mem_buf_get
  3.23      0.30     0.01       52     0.19     0.19  shadow_blur_corner
```

kisvegabor, 2020-06-26:
Great! Was measured with `lv_demo_widgets`?

embeddedt, 2020-06-26:
Yes.

embeddedt, 2020-06-26:
`lv_color_fill` probably isn't worth optimizing because it's expected to be the performance bottleneck on a system that doesn't make use of a GPU.

embeddedt, 2020-06-26:
I also did some naive "profiling" on STM32 (interrupted the processor repeatedly and noted which functions it spent more time in). Most time seems to be spent in the `_lv_style` functions and the `shadow_blur_corner` functions.

kisvegabor, 2020-06-26:
`shadow_blur_corner` is used only for the shadow of the message box. It's another story but shadow can be buffered `LV_SHADOW_CACHE_SIZE`.

Regarding the styles, let's wait @gangheivt's answer to my [comment above](https://github.com/lvgl/lvgl/issues/1605#issuecomment-650038173) and we will see how to move forward.

kisvegabor, 2020-06-30:
I was thinking about how to speed it up:
Now `lv_style_t` has a common `map` for every kind of property. We could have 4 maps for the 4 types:
- `map_opa`
- `map_int`
- `map_color`
- `map_ptr`

Only this should mean a great speed up as only 1/4 of the properties needs to be checked (only in the given type) 

Once a `map` has fixed sized elements, the properties can be stored in ascending order to perform a binary search.

embeddedt, 2020-06-30:
I like the sound of this idea. Using binary search should definitely improve performance.

kisvegabor, 2020-07-01:
In `feat/style-speed` I've implemented the separation of values (no binary search yet) and made some measurement with STM32F769. Caching was enabled and `-O3` was used. I measured the time to load the first screen of `lv_demo_widgets`. Here are the results:
```
1/48 buffer
 master: 184 ms
 new: 172 ms
 
1/20  buffer
 master: 100 ms
 new: 94 ms

1/10  buffer
 master: 67 ms
 new: 63 ms

1/8  buffer
 master: 65 ms
 new: 61 ms
 
1/20  buffer no cache
 master: 323 ms
 new no: 297 ms
```

So it's ~ 5-10% improvement on this board. Not that much... :frowning_face: 

@gangheivt Could you also try it?

kisvegabor, 2020-07-02:
I made some further tests. Created 60 small rectangles with the following style configurations and measured the rendering time in milliseconds.

| Config                  | Master                       | New                     |
|----------- |---------|----------|
| 1 style with 3 props |  15 | 15  |
| 1 style with 3 properties + 28 dummy properties [*] |  18| 17  |
| 1 style  with 28 dummy properties  + 3 properties | 18 | 17  |
| 1 style with  3 properties  + 10 empty styles|  16 | 17  |
| 11 styles with 28 + 3 properties in each | 42 | 34  |
| 11 styles 28 dummy properties only | 10 | 11 |
| 11 styles 28 dummy properties only + skip search [**] | 29 | 25 |

\* Dummy property: not used during rendering
\** Skip search: `return -1` immediately from `get_property_index`. As there are no valid properties in styles it meaks no difference in the result values.

The last 2 rows show that there is some room for optimization. It took 15-19 ms to search. However, it was an unrealistic case: 60 visible objects, and ~300 properties in each.

gangheivt, 2020-07-03:
The following is what we tested on our platform (48MHz MCU), 
1. ID description in Pictures, 
[image]
2. Compare only style access for one image 
[image]
3. Flash took ~700us access style
[image]
4. ROM took ~500us
[image]
5. LV_USE_BLEND_MODES 0 & LV_USE_OPA_SCALE 0, running in ROM, took ~450us
[image]

It seems that cache miss generate ~200us extra delay. (LV_USE_BLEND_MODES 0 & LV_USE_OPA_SCALE 0) could save about 10%. We have not tried your new style code yet.  We will run it early next week.

gangheivt, 2020-07-06:
@kisvegabor , we tested your code changes on style, the delays still similar. The major extra delays are coming from cache miss. Since our FPGA flash is slow, cache miss will have larger impact. We moved those code to RAM, it improved the performance significantly.

kisvegabor, 2020-07-06:
> We moved those code to RAM, it improved the performance significantly.

Have you modified LVGL or used `LV_ATTRIBUTE_FAST_MEM`?

gangheivt, 2020-07-06:
Yes, we tried that, that could improve the performance, reduced the extra 200us delay. It might be helpful to minimize those code in RAM.

kisvegabor, 2020-07-07:
Great!
Does it already have an acceptable performance?

gangheivt, 2020-07-07:
Yes, it could give an acceptable performance, although slower than 6.0.

kisvegabor, 2020-07-08:
A minor performance loss is acceptable for higher flexibility. 

It seems the issue can be considered solved with the use of RAM functions so I close is.

closed 2020-07-08 by kisvegabor