lvgl/lvgl #1606: Focus lost on textarea edge click

DaPa, 2020-06-23:
**Describe the bug**
1. Use LVGL 7.0.2 on lv_sim_visual_studio_sdl with lv_demo_keypad_encoder.
2. Navigate to "Text input" tab view and focus one of the two text areas.
3. Click close to the edge of the focused text area.
4. Observe that the focused text area loses the focus.
5. Click anywhere in the previously focused text area and observe that it doesn't gain the focus anymore.
6. The workaround is to focus the other text area then to come back to previous text area...

**To Reproduce**
Use existing "lv_demo_keypad_encoder()" example on lv_sim_visual_studio_sdl.

**Expected behavior**
The already focused text area doesn't loses the focus when clicked close to the edge.

**Screenshots**
[image]

[image]

**Additional context**
Seems that the text area has some child page.scrl with a border padding of 8 and indev_click_focus gets mixed up when the click goes to text area itself or to its child...
Tried to workaround indev_click_focus to check if indev_obj_act is the parent of proc->types.pointer.last_pressed (or inverse) but it's not the right way...
- Can we make the page.scrl same size as parent to cover its parent so that we don't get this "focus changing"?
- Or can we make parent "transparent" so that it never receive clicks?
- Is there a way to debug control boundaries similar to android layout bounds?
- Is there an easy way to identify each lv_obj_t: what it is and where it is in the hierarchy?

kisvegabor, 2020-06-24:
Hi,

Thanks for the detailed report!

Your suspicion was correct, It was really messed up with the scrollable - main object clicks in `indev_click_focus`

I've added a fix here: be19e91e

Could you test it?

DaPa, 2020-06-24:
Thank you for your fast response! Tested and at least the focus is not lost on edge clicking, that is good.
However now when trying to move focus on the other text area it doesn't work when clicking on the 8 pixels edge, please see capture.
[image]

PS: you probably noticed already the lv_group_get_editing CI error. I used this:
```
    else if(sign == LV_SIGNAL_FOCUS) {
#if LV_USE_GROUP
        if(lv_group_get_editing(lv_obj_get_group(obj))) {
            uint8_t state = LV_STATE_FOCUSED;
            state |= LV_STATE_EDITED;
            lv_obj_add_state(obj, state);
        }
        else 
#endif
        {
            lv_obj_add_state(obj, LV_STATE_FOCUSED);
            lv_obj_clear_state(obj, LV_STATE_EDITED);
        }
    }
```

kisvegabor, 2020-06-24:
I've fixed the CI error, thanks :)

Strange it works for me:
[image]

DaPa, 2020-06-25:
Thanks you, it works indeed! Probably I messed up the manual patching. Tried with a clean clone and works perfectly. Case can be closed

closed 2020-06-25 by DaPa

kisvegabor, 2020-06-25:
Thanks for the feedback! :slightly_smiling_face: