lvgl/lvgl #1609: Convert lvgl and lv_examples to Arduino libraries

kisvegabor, 2020-06-24:
Hi,

We already discussed the topic of converting these repos to ESP components (cc @C47D and).

The same idea was [discussed](https://github.com/lvgl/lv_arduino/issues/50#issuecomment-647360611) with @pablo2048 as well for Arduino libraries.

I've tested what are the required steps to make lvgl and lv_examples compatible with Ardunio:
- move the main header files (lvgl.h, lv_examples.h) into the src folders (too) It can contain only `#include "../lvgl.h"`
- add [library.properties](https://github.com/lvgl/lv_arduino/blob/master/library.properties) files
- add an examples folder to the lv_examples repo with an INO file, from where the examples can be called.

What do you think?

Pablo2048, 2020-06-24:
Lets to be straight - this approach require (for Arduino IDE user) to install two libraries: 1. lvgl "core" repo and 2. lv_examples repo. Actually I'm trying to tell Arduino IDE, that the lv_examples repository will be dependent on lvgl core so (maybe) the library manager will install lvgl core automatically.
There is one thing that makes me worry - we have to pay some attention that both "libraries" are in sync AKA user cannot install lvgl core 7.0.2 and lv_examples for core 6.X...

kisvegabor, 2020-06-24:
> Lets to be straight - this approach require (for Arduino IDE user) to install two libraries: 1. lvgl "core" repo and 2. lv_examples repo.

True. I need to mention that right now there is no simple way for Arduino IDE users to use the `lv_examples` repo. Compared to that this approach is big step forward. 

> Actually I'm trying to tell Arduino IDE, that the lv_examples repository will be dependent on lvgl core so (maybe) the library manager will install lvgl core automatically.

I have no experience with it but if it works, why not?

> There is one thing that makes me worry - we have to pay some attention that both "libraries" are in sync AKA user cannot install lvgl core 7.0.2 and lv_examples for core 6.X...

`lv_examples` and `lvgl` are released together so the user only needs to be sure to use the same version. It seems an acceptable thing even for novice users.

JimDrewGH, 2020-06-24:
Typically with Arduino libraries,, the "library-master" contains ALL of the files and information pertaining to whatever it is you are providing, including the copyright notice, readme, library, support files, and examples.  This is always updated as a package together (not in pieces) so that nothing ever gets out of sync.  A good example of something that is a bit non-standard, yet acceptable, is Bodmer's "TFT_eSPI".  This has the library, examples, etc. but the library uses an external config file (user_setup.h) that the user has to configure for their display type.  That user_setup.h file is located within the library's folder (which itself is located in the /libraries folder).  It would be better if this file was inside of the Arduino app (the one you are writing) folder, because every time there is an update to the TFT_eSPI library you have to go back and re-work that user_setup.h file because it is over-written as part of the update.  Hopefully with lvgl you can point any config files to app's folder instead of the library's folder, but if not then we are already dealing with that now and it's not too big of a deal.  This is the biggest downfall and confusion for everyone trying to use TFT_eSPI though, so some forethought on this might go a long ways to help eliminate technical support for LVGL.  :)

kisvegabor, 2020-06-25:
Thanks for sharing your though @JimDrewGH !

We can add a dependency to `lvgl` to automatically get the right version of `lv_examples`. It will ensure to be in sync. 

`lv_conf.h` is also located in the libraries but it's to be NOT overwritten when the library is updated. 

Maybe it'd be better to the `examples` folder into `lvgl`? This way it'd be shown automatically in PIO's package manager too. 

> but don't be surprised if you get some flak from the Adafruit crowd for not having an example using their library. :)

Adafruit already [adopted](https://learn.adafruit.com/using-littlevgl-with-adafruit-displays) LVGL, so I hope they will write :)

kisvegabor, 2020-06-28:
I've added the discussed files to lvgl and [lv_examples](https://github.com/lvgl/lv_examples). Could you take a look before publishing them?

I added a modified  [TFT_eSPI example](https://github.com/lvgl/lvgl/tree/master/examples/arduino), but in lack of hardware I couldn't test it. Can you check if it works?

Pablo2048, 2020-06-30:
Yes, I'll give it a shot and let you know...

JimDrewGH, 2020-07-03:
Are these suppose to work with LVGL for Arduino?

Pablo2048, 2020-07-03:
No, this is replacement for `lv_arduino` library. It simplifies deployment/development, but it need little bit more effort from user to make it work.

kisvegabor, 2020-07-03:
@Pablo2048 
BTW, do you have any ideas about how to use the `lv_arduino` repo? We can put any useful stuff there.

Pablo2048, 2020-07-03:
I don't think so - I thought that `lv_arduino` will be moved to 'archived' state to avoid Arduino users confusion...

JimDrewGH, 2020-07-03:
Ah, ok.  I will wait for the examples to be ported to the Arduino.

kisvegabor, 2020-07-27:
@jc78 has tested the Arduino libs (both lvgl an lv_examples) and [reported](https://forum.lvgl.io/t/touch-not-working/2770/10) that they working. So we can switch in the registry to the new lib. 

@Pablo2048 as you have experience with this can you write to Arduino about it?

kisvegabor, 2020-08-10:
Finally, I asked Arduino to add the libraries and it's done. 
See the [release note of v7.3](https://github.com/lvgl/lvgl/releases/tag/v7.3.0).

Thank you for your contribution!

closed 2020-08-10 by kisvegabor

JimDrewGH, 2020-08-13:
Unfortunately, this did not help Arduino IDE users.  When adding the example library to the Arduino IDE system (either manually using the add library function) or through the library updater, the example files are no where to be found.  They are present in the "libraries" folder, but you can't load any of the examples as a normal project because they are not shown.

Pablo2048, 2020-08-13:
Sorry guys, I'm dead man at least until the end of the next month because of my main work :-( ...

kisvegabor, 2020-08-13:
> They are present in the "libraries" folder, but you can't load any of the examples as a normal project because they are not shown.

You can (I hope it works) `#include <lv_examples.h>` and call `lv_ex_btn_1()`. So Not every example is a library, but you get a bunch of functions to call.

Is it working?

kisvegabor, 2020-08-13:
> Sorry guys, I'm dead man at least until the end of the next month because of my main work :-( ...

No worries, hang in there! :slightly_smiling_face:

JimDrewGH, 2020-08-14:
No, that doesn't work.  With the Arduino IDE, you need each of the examples in its own dedicated folder or it can't be used.  The examples can not be libraries, they are stand alone programs that can use an existing library.

Pablo2048, 2020-08-14:
The only way I can see for the Arduino IDE is to convert lv_examples to "dummy" Arduino library just with the `examples` folder IMHO...

kisvegabor, 2020-08-17:
It compiles for me
```c
#include <lvgl.h>
#include <lv_examples.h>

void setup() {
  // put your setup code here, to run once:
  lv_init();
  lv_ex_btn_1();

}

void loop() {
  // put your main code here, to run repeatedly:
}
```

Is there any problem with that?

Pablo2048, 2020-08-17:
The problem is that ArduinoIDE users can not open examples from the IDE menu like in attached image.
[image]

kisvegabor, 2020-08-17:
Thanks Pablo, I got it now.  It's really not working but I don't think it makes a big difference and reasonable:
- all examples are only 1 line of code. (call 1 function)
- we can't provide a ready to use sketch as drivers are added by the user. 

IMO once we create the [Arduino site in the docs](https://docs.lvgl.io/latest/en/html/get-started/arduino.html) and clarify this, it will be still easy to use for the users.

JimDrewGH, 2020-09-01:
Pablo is correct.  Right now, nobody using the Arduino IDE can use any of your examples.  I am just trying to get your benchmark to work under Arduino IDE and I have not been successful.

kisvegabor, 2020-09-01:
What error did you get when called `lv_demo_printer()`?

dronecz, 2020-12-06:
@kisvegabor Hi, when I try to compile `lv_demo_printer()` I get this error: 

```
C:\Users\Miroslav Zuzelka\Dropbox\@Arduino\Arduino\libraries\arduino_200877\src\lv_demo_printer\lv_demo_printer.c: In function 'lv_demo_printer':

C:\Users\Miroslav Zuzelka\Dropbox\@Arduino\Arduino\libraries\arduino_200877\src\lv_demo_printer\lv_demo_printer.c:113:41: error: 'lv_font_montserrat_22' undeclared (first use in this function)

             0, &lv_font_montserrat_14, &lv_font_montserrat_22,

C:\Users\Miroslav Zuzelka\Dropbox\@Arduino\Arduino\libraries\arduino_200877\src\lv_demo_printer\lv_demo_printer.c:113:41: note: each undeclared identifier is reported only once for each function it appears in

C:\Users\Miroslav Zuzelka\Dropbox\@Arduino\Arduino\libraries\arduino_200877\src\lv_demo_printer\lv_demo_printer.c:114:14: error: 'lv_font_montserrat_28' undeclared (first use in this function)

             &lv_font_montserrat_28, &lv_font_montserrat_32);

C:\Users\Miroslav Zuzelka\Dropbox\@Arduino\Arduino\libraries\arduino_200877\src\lv_demo_printer\lv_demo_printer.c:114:38: error: 'lv_font_montserrat_32' undeclared (first use in this function)

             &lv_font_montserrat_28, &lv_font_montserrat_32);
```

Pablo2048, 2020-12-06:
Hi Miroslav,
please check your `lv_conf.h` file (https://github.com/lvgl/lvgl/blob/b9aff011022853b13018cbe63072f44ac3367785/lv_conf_template.h#L389 , https://github.com/lvgl/lvgl/blob/b9aff011022853b13018cbe63072f44ac3367785/lv_conf_template.h#L394 ) and enable the fonts...

dronecz, 2020-12-06:
Hi @Pablo2048, 

I enabled all fonts which are mentioned in error, but error still persist. 

I guess, but can not find where, it is missing some link to that definition in that `lv_conf.h` file

embeddedt, 2020-12-06:
@dronecz Please check that you are editing the correct `lv_conf.h` file (and not the template). These definitions do work.

dronecz, 2020-12-06:
Hi @embeddedt ,

yes I´m editing right config 😉  . I can compile and upload widget example without problem 🙂 .

JimDrewGH, 2020-12-10:
How are loading the code into the IDE?  I don't see a way to do that still... other than the 3 examples from v6, there is no code available for v7.

kisvegabor, 2020-12-11:
@JimDrewGH 
Take a look at [this](https://docs.lvgl.io/latest/en/html/get-started/arduino.html#initialize-lvgl-and-run-an-example) to see how to run the examples.