lvgl/lvgl #1611: Add support of compressed fonts without prefiler

puzrin, 2020-06-25:
Currently lvgl does not supports compressed fonts without "prefilter". Font convertor supports this mode (in binary format, `--no-prefilter`), as trade-off between size/speed.

Please add this to lvgl.

kisvegabor, 2020-06-26:
As it's only an `if` to skip XOR-ing I see no problem with that.

However, we have only 2 types to determine the bitmap type: https://github.com/lvgl/lvgl/blob/274d0b522d06363ebf61228957187101bfc56de4/src/lv_font/lv_font_fmt_txt.h#L150-L153

Could you add an `LV_FONT_FMT_TXT_COMPRESSED_NO_PREFILTER` option to the lvgl writer?

puzrin, 2020-06-26:
Hm... https://github.com/lvgl/lv_font_conv/blob/master/lib/writers/lvgl/lv_table_head.js#L59

It seems, convertor already generates compression format code for this mode :). Value = 2:

```
static lv_font_fmt_txt_dsc_t font_dsc = {
    .glyph_bitmap = gylph_bitmap,
    .glyph_dsc = glyph_dsc,
    .cmaps = cmaps,
    .kern_dsc = &kern_pairs,
    .kern_scale = 16,
    .cmap_num = 1,
    .bpp = 4,
    .kern_classes = 0,
    .bitmap_format = 2
};
```

So, nothing to change in js.

kisvegabor, 2020-06-28:
True :slightly_smiling_face: 

I've added it in the `dev` branch.
ddbeb212

kisvegabor, 2020-07-08:
@puzrin 
It's already in `master`. Could you test it?

puzrin, 2020-07-08:
I did not switched to 7.x anywhere yet. Let's close as looking good.

closed 2020-07-08 by puzrin