lvgl/lvgl #1614: Rework scrolling mechanizm

kisvegabor, 2020-06-27:
Continue from https://github.com/lvgl/lvgl/issues/1539

There are some issues with the current scrolling implementation:
- Needs `lv_page`, it'd be better if it'd be supported by `lv_obj`
- Counterintuitive in some cases because scrollbars appear if an object is placed on the padding area
- Scroll propagation is limit to other page objects
- Needs an extra object (background + scrollable)
- Complex and long calculations are required to refresh layouts on a page
- Snapping is not supported by default
- Elastic "over-scroll" is not supported
- Scrolling is realized by dragging the "scrollable" part of the page which is mixing features and confuses things in some cases.

To mitigate this I came up with the following approach. 

First and foremost, separate drag and scroll. If you try to drag an object, but it has children out of its area, a scroll will be performed instead. If the object can't be scrolled further automatically (or if enabled) a parent will be searched which can be scrolled. It realizes scroll propagation. 

The object will be dragged (instead of scroll) if its explicitly enabled by `lv_obj_set_drag`.

When the object is scrolled all children are moved and the current scroll position is recorded in `scroll_x/y` variables in `lv_obj_t`.
They are used only to know where is the (0;0) coordinates of the scrolled area. (E.g. with `lv_obj_set_y(child, 0)` should calculate with `scroll_y`).

I saw that there is great support for [snapping in CSS](https://css-tricks.com/practical-css-scroll-snapping/). We can implement a subset of it.

With snapping, for example, the implementation of tabview or tileview will be much easier. 

By implementing scrolling in such a simple way would make the implementation of more advanced features much easier. (e.g. elastic scroll)

I'm still not sure if it's possible (or it's worth) to keep `lv_page` support along with the proposed "native scroll".

Let me know what you think.

embeddedt, 2020-06-27:
> The object will be dragged (instead of scroll) if its explicitly enabled by `lv_obj_set_drag`.

I think scrolling should also be driven by a copy of the CSS `overflow` property. For our purposes, `hidden`, `auto`, and `scroll` should be enough.

puzrin, 2020-06-27:
I would encourage drop of any [all] hand-made solutions in favour of CSS subset.

kisvegabor, 2020-06-29:
I agree but HTML doesn't support the "show scrollbar only when scrolled" mode which is quite common on mobile.

puzrin, 2020-06-29:
> I agree but HTML doesn't support the "show scrollbar only when scrolled" mode which is quite common on mobile.

1. I guess, if anyone wish such behaviour he can sit on scroll events and update scroller visibility.
2. Default CSS process for new extensions is add vendor-prefixed style properties. If it's absolutely impossible to avoid, this approach can be used in lvgl too.

embeddedt, 2020-06-29:
> I agree but HTML doesn't support the "show scrollbar only when scrolled" mode which is quite common on mobile.

Mobile browsers diverge from desktop browsers in this regard, and usually don't show the scrollbar except when scrolling.

kisvegabor, 2020-06-29:
@puzrin 
> Default CSS process for new extensions is add vendor-prefixed style properties. If it's absolutely impossible to avoid, this approach can be used in lvgl too.

How do you imagine it in the practice? If we have the "standard" `LV_STYLE_OVERFLOW_SCROLL` how to a name a custom `FOO` mode?

@embeddedt 
> Mobile browsers diverge from desktop browsers in this regard, and usually don't show the scrollbar except when scrolling.

So what do you suggest? Are we free to add custom options?

puzrin, 2020-06-29:
> How do you imagine it in the practice? If we have the "standard" `LV_STYLE_OVERFLOW_SCROLL` how to a name a custom `FOO` mode?

https://developer.mozilla.org/en-US/docs/Web/CSS/::-webkit-scrollbar

Something like `[::webkit-]scrollbar-hide-inactive`

kisvegabor, 2020-06-30:
I see but how to name it LVGL? Everything is prefixed with `LV_` + `STYLE`. I mean I can't imagine a "good looking" way to name extended functionalities with the naming conventions of LVGL.

embeddedt, 2020-06-30:
Is there a particular feature missing from CSS `overflow`? I don't see the need for any additional functionality beyond what CSS provides.

puzrin, 2020-06-30:
This question is ortogonal to more important - data structs & api signatures. It should not lock anything. Just name it somehow temporary and move forward. You can rename it later.

kisvegabor, 2020-06-30:
@embeddedt 
>  I don't see the need for any additional functionality beyond what CSS provides.

Than hoe to handle to "auto hide" feature?

@puzrin 
> This question is ortogonal to more important - data structs & api signatures. It should not lock anything. Just name it somehow temporary and move forward. You can rename it later.

I was asking it because you mentioned "vendor-specific" things should be prefixed and I couldn't imagine a right way to do it. 

----

To move forward with the HTML/CSS compatibility topic we need to agree on to what extent shall we stick with HTML/CSS.

IMO as LVGL already quite different from HTML/CSS in principle so there is no reason for a very close compatibility. We can copy what seems reasonable but can't make users' life more difficult just to ensure compatibility. 

The compatibility would be important if one wanted to create a browser with LVGL but it's not the purpose of this library.

puzrin, 2020-06-30:
> Is there a particular feature missing from CSS `overflow`? I don't see the need for any additional functionality beyond what CSS provides.

Yes, it's not needed in scope of lvgl use cases IMO. At least, i would suggest to postpone it to future iterations.

kisvegabor, 2020-06-30:
> Yes, it's not needed in scope of lvgl use cases IMO. At least, i would suggest to postpone it to future iterations.

I don't agree. The UI's created with LVGL are rather mobile-like then desktop. And on mobile it's a common thing.

puzrin, 2020-06-30:
> I was asking it because you mentioned "vendor-specific" things should be prefixed and I couldn't imagine a right way to do it.

I had a bit different focus in that context. How CSS is extended with not-existing features. Vendors add new CSS class properties, not additional objects as `page` has now.

Special prefixing is not critical, if you have few of those.

> I don't agree. The UI's created with LVGL are rather mobile-like then desktop. And on mobile it's a common thing.

Could you show me an example image of interface where `overflow` needed?

embeddedt, 2020-06-30:
> Than how to handle to "auto hide" feature?

I suggest automatically hiding the scrollbar with `overflow: auto`, and always showing it with `overflow: scroll`. I think that's similar enough to how mobile browsers work.

puzrin, 2020-06-30:
> IMO as LVGL already quite different from HTML/CSS in principle so there is no reason for a very close compatibility. We can copy what seems reasonable but can't make users' life more difficult just to ensure compatibility.

IMO you try to compare "not existing [but 100% good] imaginated" with "well known existing". If you argue to something imaginated, it has "very uncertain and risky" satus. And of cause, considering it as ideal is logical mistake [but very common for humans]. 

Partial copy has almost the same cost as reinvent everything. But it's your time, and nobody can prohibit you spend much more for your own way if you wish. I can only explain cost's difference and risks on the way to "something beautiful and stable".

embeddedt, 2020-06-30:
> IMO as LVGL already quite different from HTML/CSS in principle so there is no reason for a very close compatibility.

@kisvegabor I'd actually say that they've become more similar than one might think at first glance. We have the equivalent of HTML elements (widgets), a CSS equivalent (the style system) and custom object-specific attributes (essentially the same as HTML). The main differences at this point are the syntax and the fact that functionalities like layout and scrolling are embedded in their own objects.

kisvegabor, 2020-06-30:
> I suggest automatically hiding the scrollbar with overflow: auto, and always showing it with overflow: scroll. I think that's similar enough to how mobile browsers work.

In HTML `auto` is to show only when necessary but nothing about auto-hide. So:
- `scroll` always show
- `auto` show when there is something to scroll
- `hidden` never show
- `???` show only when scrolled

> I'd actually say that they've become more similar than one might think at first glance. We have the equivalent of HTML elements (widgets), a CSS equivalent (the style system) and custom [...]

I agree to go to that direction but I think we shouldn't "fear" to shape it a little. Maybe the thing is to define what is accepted when "shaping":
- ok to use a subset of features (e.g. overflow, flexbox)
- ok to add custom (independent) properties (e.g. overflow: auto-hide)
- avoid reinventing features if there are similar and usable (in LVGL's case) HTML/CSS alternative

What do you think?

embeddedt, 2020-06-30:
Auto-hiding seems to be a vendor specific choice. Mobile browsers tend to auto-hide the scrollbar while desktop browsers always show it.

I think your definition of shaping is fine.

puzrin, 2020-06-30:
> * ...
> * ok to add custom (independent) properties (e.g. overflow: auto-hide)
> * ...
>
> What do you think?

The only question is how often do you plan to use that case. Because, technically, everything can be expressed with it. It's important to keep in mind, this choice is possible as exclusion, but not welcome. And should be used only after you tried to apply all other alternatives.

In this case "subset of features" will really be subset, not "partial copy with modifications".

kisvegabor, 2020-06-30:
> The only question is how often do you plan to use that case. Because, technically, everything can be expressed with it. It's important to keep in mind, this choice is possible as exclusion, but not welcome. And should be used only after you tried to apply all other alternatives.

Okay, let's consider it as the last option.

kisvegabor, 2020-07-02:
I started to work on scrolling.
Here is a test with HTML to see how it works in non-trivial cases:
[index.html.zip](https://github.com/lvgl/lvgl/files/4863376/index.html.zip)
[image]

Things to note:
- Can't be scrolled to objects out on on the left or top
- Objects on the padding area don't make the parent scrollable  (blue). But there is a bug (?) in Chrome if the horizontal scrollbar appears the parent can be scrolled vertically a little (the height of the horizontal scrollbar?) but no vertical scrollbar appears. Just can be scrolled. It works well in Firefox. 
- Can be scrolled to the object out of the parent's box. However, the padding of the parent is not respected, the object is shown on the very edge of the parent. (lime)

I'm ok with doing the same in LVGL. 

---

Here is what I've done in LVGL:
- Scroll feature for the basic object. (screen in the example)
- Elastic scroll if scrolled inside
- Jump back on release if scrolled inside.

[image]

kisvegabor, 2020-07-08:
Update on what is already working:
- elastic scroll
- scroll chain
- scrollbar (not styled)
- only vertical scroll

It's a little bit hacky here and there but conceptually all interesting features are working, 

[image]

embeddedt, 2020-07-08:
Looks very good, @kisvegabor!

kisvegabor, 2020-07-21:
Update:
- The followings were already working: elastic scroll, scroll chaining
- Add support `on`, `off`, `auto` and `active` modes
- Scrolling can happen only in one direction at once (no x + y at the same time). "Both axis mode" can be added later if required.
- Remove `drag`. It was added earlier mainly to handle scrolling. Something like this can be simply achieved by setting the position in the PRESSING event.
- Scrollbar styling was added as [direct style properties](https://github.com/lvgl/lvgl/blob/7a1d9e77538a8d3253143fe0fbbcdf5bc7b15a44/src/lv_core/lv_style.h#L201-L209) instead of a pseudo-class. It keeps the base object lightweight. 
- Snaping is still missing.

I haven't updated all the widgets yet. It seems more reasonable to update with the new layout at once. (Base obj, label, container, button are working for testing)

I was testing with this code:
```c
  lv_obj_t * p0 = lv_obj_create(lv_scr_act(), NULL);
  lv_obj_set_size(p0, 150 ,150);
  lv_obj_set_pos(p0, 10, 20);
  lv_obj_t * l = lv_label_create(p0, NULL);
  lv_label_set_text(l, "abcdefabcdefabcdefabcdefabcdefabcdefabcdefabcdefabcdefabcdefabcdefabcdef0\na\na\na\na\na\na\na\na\na\naX");

  lv_obj_t * btn = lv_btn_create(p0, NULL);
  lv_obj_set_pos(btn, 130, 30);

  lv_obj_t * p = lv_obj_create(p0, NULL);
  lv_obj_set_size(p, 100 ,100);
  lv_obj_set_pos(p, 10, 20);
  l = lv_label_create(p, NULL);
  lv_label_set_text(l, "a\nb\nc\nd\ne\nf\na\nb\nc\nd\ne\nf"
          "\na\nb\nc\nd\ne\nf\na\nb\nc\nd\ne\nf\n0");
```

Feedback is welcome. :slightly_smiling_face:

puzrin, 2020-07-22:
When scrolling and new layouts are planned for landing? I still sit on 6.x, and would skip 7.x if 8.x come soon.

kisvegabor, 2020-07-23:
As far as I can see now, these two things will be the leading new feature of v8. 
For the release, I'd say September/October.

incity, 2020-08-06:
could you also think about to make all object with the transformation(rotate scale ..), like this:
[](https://www.w3school.com.cn/tiy/t.asp?f=css3_transform_rotate)

kisvegabor, 2020-08-08:
@incity 
It's not on the roadmap now. 
Buttons, text, etc are drawn on the fly and NOT rendered to images first. It'd be quite complicated to make a transformation on a non-rasterized (rendered) object.

excitedbox, 2020-08-10:
That is really neat. That little bounce in the scroll does so much for giving the app character even though it is such a small thing.

I held off waiting for v7 to be released but seeing the recent work that you have done makes me want to wait for v8.

I would have to check but some of the things you mentioned about how the HTML implementation handles it could probably be down to the browser or using off canvas stuff could fix some of the snapping. I believe that padding is not counted as part of the object so it is ignored on the lime colored square.

To the question of how closely to stick to CSS/HTML standards I think it is best to only adhere to it as closely as makes sense. If there is something that is better for LVGL to change then I think it is better to change it. I really think the only thing that should be imitated is the Style syntax but anything else can be changed. CSS and HTML are very resource intensive and a lot of MCUs lack the power for the fancy features. There is a reason that there are no good CSS, JS and HTML engines for MCUs and even projects like Espruino, Ultralight, and others have only a subset of the standards implemented. I think what makes LVGL so great is how many awesome features you have managed to pack in while keeping it so light.

kisvegabor, 2020-08-11:
> That is really neat. That little bounce in the scroll does so much for giving the app character even though it is such a small thing.

Happy to hear that you like it! :slightly_smiling_face: 

>  I believe that padding is not counted as part of the object so it is ignored on the lime colored square.

Do you know what is the rationale behind it? I'd expect to keep the padding.

> To the question of how closely to stick to CSS/HTML standards I think it is best to only adhere to it as closely as makes sense. 

I agree. However, it turned out "as closely as makes sense" means different for everybody. :smile:

excitedbox, 2020-08-13:
Padding is there to increase the hitbox for small items so it really should count but in text docs margins are not part of the text area. Keep in mind HTML and CSS were not designed to be used the way they are now. Originally HTML was meant for just displaying text/image content and CSS was actually supposed to let users style that content to fit their liking. Padding was added to allow adding more space around images and text while margin was to add space around tables, content areas etc. Now they are both used to help create entire website UIs. The entire web is just hacked together really with features being added as needed to replace those hacks. Tables layouts for instance being replaced with divs and layers is a prime example. This is why browser support is so shitty for various features. That is also another reason it is more important to focus on user expectation than to copy a specific tech. Blender is a pain to use because the control scheme is totally different than what you are used to from every other graphics software. Every software except blender deselects an object when clicking the background for instance. That causes missclicks and frustration.

PS.
JS was written in about 10 days to give basic interactivity without having to use Active X or Java Applets. I remember back in the day it was slow as hell and caused stuttering in basic image galleries. Crazy how far it has come. My first JS script was a film strip style image scroller that changed the description of paintings the website of a local Art Gallery website I was making and it was so amazing for 2003 technology. Shortly after lightbox case out and caused lag spikes on some computers.

kisvegabor, 2020-08-13:
> The entire web is just hacked together really with features being added as needed to replace those hacks. Tables layouts for instance being replaced with divs and layers is a prime example. 

Good to know that not I'm the only person who feels this way. :laughing: 

I read some articles about "why people feel that CSS is frustrating" and all of them said because people don't make the effort to correctly learn it. It seems you can't know CSS a little. I don't even need to know how to write functions to play with `C` (see Arduino),  but I need to know a large slice of CSS to do anything. It should be clearly avoided in LVGL. 

Anyway, there are some battle-tested concepts in it and its really good starting point. I believe by carefully picking features we can adopt what people like in CSS/HTML and create a familiar environment for them. 

> JS was written in about 10 days to give basic interactivity without having to use Active X or Java Applets. 

I was 5 years old back then. :smile:

kisvegabor, 2020-09-06:
A quick report about what is working now:
- elastic scroll
- scroll chaining (out of the box, detects if a parent can be scrolled)
- scroll momentum (throw)
- scroll bar modes (on, off, active (being scrolled), auto (has overflow))
- scrollbar styling
- `lv_obj_set/get_pos` takes scroll position into account (y = 10 is 10 independently from scroll position)
- snapping children to parent start/end/center (can be different on x/y axis)
- snap-stop to allow scrolling only 1 snap point
- disable scrolling in any direction (to create something like in tileview)

Some API polishing is still required but it's very close...

kisvegabor, 2020-09-23:
I think all discussed regarding scrolling and the current state is available in the `dev-v8` branch so I close this issue.

Thank you very much for the discussion!

closed 2020-09-23 by kisvegabor