lvgl/lvgl #1617: STM32 DMA2D definitions conflict with STM32Cube

embeddedt, 2020-06-27:
The [DMA2D color formats declared in `lv_gpu_stm32_dma2d.h`](https://github.com/lvgl/lvgl/blob/3e8b39c4046cff57a9a211d0b572cdad74344f3f/src/lv_gpu/lv_gpu_stm32_dma2d.h#L23-L27) have the same name (and value) as those provided by the Cube library. This causes redefinition errors at compile time.

I think the best solution here is to simply prepend `LV_` to the names so they are in their own namespace.

@kisvegabor If you approve of this change I'll apply it to `master`.

(cc @microwavesafe)

microwavesafe, 2020-06-28:
Sounds like a good idea to me. We actually don't need all those defines only a few are compatible with lvgl.

kisvegabor, 2020-06-28:
I'm also okay with that.

embeddedt, 2020-06-28:
Great; I've just applied the change.

commit 421f1b2c (2020-06-28): lv_gpu: fix #1617 by renaming internal macros

This avoids conflicts with STM32Cube.
closed 2020-06-28 by embeddedt