lvgl/lvgl #1618: chinese word use underline function

juice-man, 2020-06-28:
# It seem too close to the word ,when I use the underline function in chinese word.
for example:
[image]

liangyongxiang, 2020-06-28:
What font do you use?
Can you test the characters g and |?

juice-man, 2020-06-28:
[image]

[chinese_ui.zip](https://github.com/lvgl/lvgl/files/4842129/chinese_ui.zip)

attachment is test result and font

liangyongxiang, 2020-06-28:
How are your font information generated and what tools are used?
You need to provide a minimal example to reproduce.

In addition, you can use font editing software such as [fontforge](https://fontforge.org/en-US/) to open your font file. The following is an example:

[image]

[image]

From [the code here](https://github.com/lvgl/lvgl/blob/41e65d9df1f45c7ecf296913190fd10eccaf6138/src/lv_draw/lv_draw_label.c#L345), you can see how the coordinates of the underline are calculated.

I think your underline is drawn on the baseline. You can try to modify the value of baseline, but I have not tested it.

liangyongxiang, 2020-06-28:
I think the result you need is `text-underline-position: under;`

https://developer.mozilla.org/en-US/docs/Web/CSS/text-underline-position

under:
[image]

auto:
[image]

incity, 2020-06-29:
> How are your font information generated and what tools are used?
> You need to provide a minimal example to reproduce.
>
> In addition, you can use font editing software such as [...]

I think the problem is the underline and the font glyph  are all drawn start from baseline. @kisvegabor

juice-man, 2020-06-29:
I used freetype2 to generate font information and call "lv_style_set_text_decor(&style, LV_STATE_DEFAULT, LV_TEXT_DECOR_UNDERLINE);" set underline function。

the follow is my code:

`

    lv_freetype_init(64); 
    static lv_font_t font1;
    lv_freetype_font_init(&font1, "/system/res/font/chinese_ui.ttf", 22);
    
    /*Create style with the new font*/
    static lv_style_t style;
    lv_style_init(&style);
    lv_style_set_text_font(&style, LV_STATE_DEFAULT, &font1);
    lv_style_set_text_decor(&style, LV_STATE_DEFAULT, LV_TEXT_DECOR_UNDERLINE);

    /*Create a label with the new style*/
    lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
    lv_obj_add_style(label, LV_LABEL_PART_MAIN, &style);
    lv_obj_set_y(label, 100);
    //lv_label_set_text(label, "d");
    lv_label_set_text(label, "中文 world and g and |");
    lv_obj_set_drag(label, 1);
    lv_obj_add_protect(label,LV_PROTECT_PRESS_LOST);
`

kisvegabor, 2020-06-29:
The underline is really drawn on the baseline of the font. The width is automatically calculated as `height / 10` and it's symmetrically drawn on the baseline. 

Probably it'd be better to draw it below the baseline (i.e. the top of the underline should be on the baseline) 

However, the best would be to extract it from the font. FreeType can tell it like [this](https://www.freetype.org/freetype2/docs/tutorial/step2.html).

@puzirin Can we add this information to the font description?

kisvegabor, 2020-06-29:
cc @puzrin

puzrin, 2020-06-29:
Sure. Please create issue with description, what should be generated in lvgl format.

kisvegabor, 2020-07-08:
Let's continue in lvgl/lv_font_conv#40

closed 2020-07-08 by kisvegabor