lvgl/lvgl #1621: Checkbox can't be enabled after disabled

weixiongmei, 2020-06-29:
**Describe the bug**

The checkbox can't be re-enabled after it's disabled, because the limitation of the API in LVGL V7.0.2

```C
/**
 * Make the check box inactive (disabled)
 * @param cb pointer to a check box object
 */
static inline void lv_checkbox_set_disabled(lv_obj_t * cb)
{
    lv_btn_set_state(cb, LV_BTN_STATE_DISABLED);
}
```

**To Reproduce**

**Expected behavior**

The API lv_checkbox_set_disabled can't pass a second param to re-enable the checkbox after disabled.

pete-pjb, 2020-06-29:
Hi @weixiongmei,

Looking at the API, you should be able to call:

```c
lv_checkbox_set_checked(your_ch_box, false);
```

To re-enable the widget.

Kind Regards,

Pete

weixiongmei, 2020-06-29:
@pete-pjb not to re-enable it, want to be checked & disabled at the same time, it can only checked **or** disabled, but can't be both.

embeddedt, 2020-06-29:
I think a workaround in the meantime is to use `lv_btn_set_state(cb, LV_BTN_STATE_CHECKED_DISABLED)`.

kisvegabor, 2020-06-29:
> I think a workaround in the meantime is to use lv_btn_set_state(cb, LV_BTN_STATE_CHECKED_DISABLED).

There really should be an `lv_checkbox_set_state(cb, LV_BTN_STATE...)` function.

@weixiongmei 
You can use the genereic `lv_obj_add/clear_state(cb, LV_STATE_DISABLED)` functions.

pete-pjb, 2020-06-30:
Hi @kisvegabor ,

Would you like me to add the 

> lv_checkbox_set_state(cb, LV_BTN_STATE...) function.

If so which repo would you like me to add it to?

Kind Regards,

Pete

kisvegabor, 2020-06-30:
> Would you like me to add the

It'd be great! Thanks!

> If so which repo would you like me to add it to?

Do you mean "which branch"? If so, it can go to the dev branch, as it's a new feature.

pete-pjb, 2020-07-01:
Sorry yes I meant which branch, I am doing this now...

I have also fixed the function  ```lv_theme_set_base()``` renamed the variable clashing with c++ keyword 'new' to 'new_theme' with this PR, will send later...

pete-pjb, 2020-07-01:
I will close this now have added new functions here: #1630

closed 2020-07-01 by pete-pjb