lvgl/lvgl #1623: LV_USE_LOG custom log function type error

k3wals, 2020-06-29:
When registering a custom log function per the example in the [documentation](https://docs.lvgl.io/en/html/porting/log.html), there is a type mismatch. Expects it to be of type `lv_log_print_g_cb_t`.

**To Reproduce**
Custom function (before main)
```c
#if LV_USE_LOG != 0
  void my_print(lv_log_level_t level, const char * file, int line, const char * fn_name, const char * dsc)
  {
    /*Send the logs via serial port*/
    if(level == LV_LOG_LEVEL_ERROR) NRF_LOG_INFO("ERROR: ");
    if(level == LV_LOG_LEVEL_WARN)  NRF_LOG_INFO("WARNING: ");
    if(level == LV_LOG_LEVEL_INFO)  NRF_LOG_INFO("INFO: ");
    if(level == LV_LOG_LEVEL_TRACE) NRF_LOG_INFO("TRACE: ");

    NRF_LOG_INFO("%s@%d->%s\r\n", file, line, dsc);
  }
#endif
```
Register function (in main)
```c
 NRF_LOG_INFO("LVGL Init");
  lv_init();
  #if LV_USE_LOG != 0
    lv_log_register_print_cb(my_print); //register custom lvgl logger function
  #endif
```

Compilation error
```
PS C:\Users\Username\Documents\nrf_project_v1\project> make -j4
Compiling file: main.c
Compiling file: st7789.c
Compiling file: lv_font_unscii_8.c
Compiling file: lv_font_montserrat_46.c
Compiling file: lv_font_montserrat_38.c
Compiling file: lv_font_montserrat_26.c
Compiling file: lv_font_montserrat_16.c
./src/main.c: In function 'main':
./src/main.c:820:30: error: passing argument 1 of 'lv_log_register_print_cb' from incompatible pointer type [-Werror=incompatible-pointer-types]
     lv_log_register_print_cb(my_print); //register custom lvgl logger function       
                              ^~~~~~~~
In file included from ./lvgl/lvgl.h:17,
                 from ./inc/st7789.h:9,
                 from ./src/main.c:34:
./lvgl/src/lv_misc/lv_log.h:62:51: note: expected 'lv_log_print_g_cb_t' {aka 'void (*)(signed char,  const char *, long unsigned int,  const char *, const char *)'} but argument is of type 'void (*)(lv_log_level_t,  const char *, int,  const char *, const char *)' {aka 'void (*)(signed char,  const char *, int,  const char *, const char *)'} 
 void lv_log_register_print_cb(lv_log_print_g_cb_t print_cb);
                               ~~~~~~~~~~~~~~~~~~~~^~~~~~~~
Compiling file: lv_font_montserrat_12.c
Compiling file: lv_font_simsun_16_cjk.c
Compiling file: lv_font_montserrat_12_subpx.c
cc1.exe: all warnings being treated as errors
make: *** [../nrf-sdk/components/toolchain/gcc/Makefile.common:272: _build/adc_sensor/main.c.o] Error 1
make: *** Waiting for unfinished jobs....
```

**Expected behavior**

Works per documentation example.

**Screenshots**
n/a

**Additional context**

n/a

embeddedt, 2020-06-29:
From the error, it looks like the function signature should be `uint32_t line` and not `int line`. Can you confirm that that works without errors?

kisvegabor, 2020-06-29:
I couldn't find what is the type of `_LINE_`. If you know it's `int` we can change the prototype.

k3wals, 2020-06-29:
Changed the function (int line to uint32_t line)
```c 
static lv_log_print_g_cb_t my_print(lv_log_level_t level, const char * file, uint32_t line, const char * fn_name, const char * dsc)
```

Getting following compile error. These are not straightforward to interpret at all!

```Compiling file: main.c
./src/main.c: In function 'main':
./src/main.c:808:30: error: passing argument 1 of 'lv_log_register_print_cb' from incompatible pointer type [-Werror=incompatible-pointer-types]
     lv_log_register_print_cb(my_print); //register custom lvgl logger function
                              ^~~~~~~~
In file included from ./lvgl/lvgl.h:17:0,
                 from ./inc/st7789.h:9,
                 from ./src/main.c:34:
./lvgl/src/lv_misc/lv_log.h:62:6: note: expected 'lv_log_print_g_cb_t {aka void (*)(signed char,  const char *, long unsigned int,  const char *, const char *)}' but argument is of type 'void (* (*)(lv_log_level_t,  const
char *, uint32_t,  const char *, const char *))(lv_log_level_t,  const char *, uint32_t,  const char *, const char *) {aka void (* (*)(signed char,
 const char *, long unsigned int,  const char *, const char *))(signed char,  const char *, long unsigned int,  const char *, const char *)}'
 void lv_log_register_print_cb(lv_log_print_g_cb_t print_cb);
      ^~~~~~~~~~~~~~~~~~~~~~~~
cc1.exe: all warnings being treated as errors
make: *** [../nrf-sdk/components/toolchain/gcc/Makefile.common:272: _build/adc_sensor/main.c.o] Error 1
```

kisvegabor, 2020-06-30:
Great! I close this issue then.

closed 2020-06-30 by kisvegabor