lvgl/lvgl #1624: lv_textarea_add_char fails on big endian hardware

jbamaral, 2020-06-30:
**Describe the bug**

`lv_textarea_add_char` takes an `uint32_t` char parameter (in order to support wide chars) but it casts the parameter directly to a `char *` assuming little endian byte order.

This makes it fail on big endian scenarios.

**To Reproduce**

In a big endian hardware use the `lv_widgets_demo` application and try to press the line feed key on the keyboard. It will issue an `lv_textarea_add_char` call with `'\n'` as parameter and will fail as it will pass zero as the first argument of the casted buffer.

**Expected behavior**

Line feed button to work.

**Additional context**

Found the bug while testing [with this terminal](https://www.verifone.com/en/us/devices/portables-transportables/vx-690)

kisvegabor, 2020-06-30:
Modifying  the beginning of lv_textarea_add_char like this could solve it:
```c
#if LV_BIG_ENDIAN_SYSTEM == 0
    c = _lv_txt_encoded_conv_wc(c);
#endif

    char letter_buf[5] = {0};
    letter_buf[0] = (char)(c&0xFF);
    letter_buf[1] = (char)((c>>8)&0xFF);
    letter_buf[2] = (char)((c>>16)&0xFF);
    letter_buf[3] = (char)((c>>24)&0xFF);
    letter_buf[4] = '\0';
```
Can you test it?

jbamaral, 2020-06-30:
> Modifying the beginning of lv_textarea_add_char like this could solve it:
>
> ```c
> #if LV_BIG_ENDIAN_SYSTEM == 0
>     c = _lv_txt_encoded_conv_wc(c);
> #endif
>
>     char letter_buf[5] = {0}; [...]

I'll test it here.

jbamaral, 2020-06-30:
I'm about to start testing although I have a question here. The comment on the function says `* To add a wide char, e.g. 'Á' use lv_txt_encoded_conv_wc('Á')`

Are you willing to change the behavior to accept directly 'Á'? Is that why you are converting before adding to the array? Wouldn't that change broke existing apps?

kisvegabor, 2020-06-30:
> Are you willing to change the behavior to accept directly 'Á'? Is that why you are converting before adding to the array? Wouldn't that change broke existing apps?

That's true. I'm still thinking about what to do because it'd be quite clean to integrate `lv_txt_encoded_conv_wc` in `lv_textarea_add_char`. It's a very marginal thing so maybe we can make an exception here to modify the API without a major release. 

Or another approach could be to return the letter as it is from `lv_txt_encoded_conv_wc` because it's already correct on big-endian systems.

@embeddedt What do you think?

embeddedt, 2020-06-30:
If there's no need to change the letter in `lv_txt_encoded_conv_wc` then I suggest we just return it. It makes the most sense to me and should fix the root problem of big-endian yielding different results.

kisvegabor, 2020-06-30:
Okay, let's do this way.

jbamaral, 2020-06-30:
Ok, after playig for a while with the example I've found that the problem can be solved withtout changing the callback as follows:

```c
    char *letter_buf;

    uint32_t u32_buf[2];
    u32_buf[0] = c;
    u32_buf[1] = 0;
    
    letter_buf = (char*)&u32_buf;

#if LV_BIG_ENDIAN_SYSTEM
    while (*letter_buf == 0) ++letter_buf;
#endif
```

The reason I'm suggesting it is that the callback could be to other formats different than utf8 and it's best if we solve the problem locally. It works because after converting the byte order is ok on big endian but in memory it puts the zeroes first. Please let me know if I should push this and if not please point me to the right thing to do. 

My test code was:
```
void
simpleDemo(void)
{
    lv_obj_t * ta1 = lv_textarea_create(lv_scr_act(), NULL);
    lv_obj_set_size(ta1, 230, 200);
    lv_obj_align(ta1, NULL, LV_ALIGN_CENTER, 0, 0);
    lv_obj_set_style_local_text_font(ta1,
                                     LV_TEXTAREA_PART_BG,
                                     LV_STATE_DEFAULT,
                                     &lv_font_dejavu_16_persian_hebrew);
    lv_textarea_set_text(ta1, "1. 12 ئ \n");                                     
    lv_textarea_add_text(ta1, "2. 1 á 2 Á ab ئ \n");
    lv_textarea_add_text(ta1, "3. 12 ئ ");
    lv_textarea_add_text(ta1, "ئ");
    lv_textarea_add_text(ta1, "34 \n");
    
    lv_textarea_add_char(ta1, '4');
    lv_textarea_add_char(ta1, '.');
    lv_textarea_add_char(ta1, ' ');
    lv_textarea_add_char(ta1, _lv_txt_encoded_conv_wc(0xD8A6));
    lv_textarea_add_char(ta1, 'a');
    lv_textarea_add_char(ta1, ' ');
    lv_textarea_add_char(ta1, _lv_txt_encoded_conv_wc(0xD8A6));
    lv_textarea_add_char(ta1, ' ');
    lv_textarea_add_char(ta1, 'B');
    
}
```

jbamaral, 2020-06-30:
Humm, maybe just shift bytes in lv_txt_uft8_conv_wc will be better... Because if we use another formatting then it's no use to actually trying to guess... I'll change the lv_txt_utf8_conv_wc to shift and check if the resulting original code works.

jbamaral, 2020-06-30:
Ok. Skipping in `lv_txt_utf8_conv_wc` alone does not work because `lv_textarea_add_char(`a`)` must work without calling `_lv_txt_encoded_conv_wc`. 

Therefore I rebase and force pushed a solution that I think it's acceptable for a hotfix in #1620 

Please review and advise. Thanks!

jbamaral, 2020-07-01:
I'm closing this issue as it's being solved by #1620

closed 2020-07-01 by jbamaral