lvgl/lvgl #1629: DMA2D hardfault with RGB565 format

kisvegabor, 2020-07-01:
**Describe the bug**

On STM32F769 Discovery with DMA2D enabled when I set `LV_COLOR_DEPTH 16` hard fault occurs. (32 bit color depth work fine) The global variables re overwritten to `FF` so I suppose it's and bounds write issue.

**To Reproduce**

ON ST32F769 select `LV_COLOR_DEPTH 16`, enable `LV_USE_GPU_STM32_DMA2D` and start the `lv_demo_widgets()`. It crashes before showing teh first screen.

**Additional context**

It works with the old HAL dependent DMA2D driver.
`git checkout ee6c5e33 src/lv_gpu/`

@springltd Could you take a look?

microwavesafe, 2020-07-01:
Interesting I use colour depth 16, but on the F429 and this works fine with the demo_widgets.

microwavesafe, 2020-07-01:
My guess would be the DMA2D is writing over the wrong RAM location? Did the HAL do something with the target location? When we set up the colour format, it dictates the number of actual bytes written by the DMA2D, if we pass in the correct target location and have set the right colour depth, there's nothing else to control.

I can't actually replicate this here, can you step through and see what function is crashing and whether the location and colour format settings look correct?

microwavesafe, 2020-07-01:
I did read as well that the target and source locations must be aligned based on the colour format. If you look at the description of the memory location registers (e.g. DMA2D_OMAR), you get this.

`The address alignment must match the image format selected e.g. a 32-bit per pixel
format must be 32-bit aligned and a 16-bit per pixel format must be 16-bit aligned`

Is it possible that your source or target locations are misaligned and this is causing the hard fault?

embeddedt, 2020-07-01:
I'm also not able to reproduce this with 16-bit color on STM32F746 using the new GPU driver. My buffer configuration is pasted below:

```c
        static lv_disp_buf_t disp_buf_1;

	static lv_color_t buf1_1[LV_HOR_RES_MAX * 68];                      /*A buffer for 10 rows*/
	static lv_color_t buf1_2[LV_HOR_RES_MAX * 68];                      /*A buffer for 10 rows*/
	lv_disp_buf_init(&disp_buf_1, buf1_1, buf1_2, LV_HOR_RES_MAX * 68);   /*Initialize the display buffer*/
```

kisvegabor, 2020-07-01:
Thank you for looking into it.

I've just found the issue. My TFT driver was also initialized DMA2D and reset it. (Default ARGB8888)
The fix is [here](https://github.com/lvgl/lv_port_stm32f769_disco/commit/528fdaaffa7b4fdc1d910a50defdf0382e26006d).
The original `weak` function with the offending line is [here](https://github.com/lvgl/lv_port_stm32f769_disco/blob/master/Utilities/STM32F769I-Discovery/stm32f769i_discovery_lcd.c#L1703).

closed 2020-07-01 by kisvegabor