lvgl/lvgl #1631: there is a problem with lv_img's click area after transforming.

incity, 2020-07-02:
1. create 2 widgets and set them draggable : lv_label first and then lv_img. (i.e. lv_label is behind lv_img)
2. change the size of lv_img or rotate it.
3. press the image to drag, but it will drag the label instead. (label is behind the image).

embeddedt, 2020-07-02:
This behavior is by design. Image rotation [does not affect input processing](https://forum.lvgl.io/t/how-to-update-the-coords-when-the-image-is-scaled/1464/2?u=embeddedt).

incity, 2020-07-02:
@embeddedt before changing the size of the image, it can be dragged normally. but after changing the size on runtime, the image becomes hard to drag.

embeddedt, 2020-07-02:
Can you provide a code sample?

incity, 2020-07-02:
@embeddedt  just call lv_obj_set_size

kisvegabor, 2020-07-02:
Please create a short video about the issue and 
- send code example with the label and image creation and settings. 
- attack the converted image

incity, 2020-07-03:
@kisvegabor @embeddedt base on the lv_demo_printer.c 👍 
[image]

    lv_img_set_pivot(scan_img, 0, 0); // seems this cause the problem
    lv_img_set_zoom(scan_img, 512);
    lv_obj_set_drag(scan_img, true);

kisvegabor, 2020-07-03:
I see. I improved it in `master` here: e9d3001d

closed 2020-07-03 by incity