lvgl/lvgl #1632: How to adjust the height of the tabview?(v7.xxx)

LusterWong, 2020-07-02:
# All enhancement requests or questions should be directed to the Forum.

LVGL was my first use and I lacked experience. Now I'm using the latest V7.0.2 and running emulation on a PC.

Since tabview required, and the display device resolution only 240x240, although the font of the tabview can be set, how should the height be adjusted? for example I'm using a 12-point font, but I want my TabView to be 20 pixels high. What should be done?

Thx.

We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

kisvegabor, 2020-07-02:
See here: https://docs.lvgl.io/latest/en/html/widgets/tabview.html#parts-and-styles

---

The title from the issue template says: "All enhancement requests or questions should be directed to the Forum."

So I close this issue. If you have further question please ask them in the forum:
http://forum.lvgl.io/

LusterWong, 2020-07-02:
> See here: https://docs.lvgl.io/latest/en/html/widgets/tabview.html#parts-and-styles
>
> The title from the issue template says: "All enhancement requests or questions should be directed to the Forum." [...]

Sorry, I was too hasty. I'm not talking about the height of the tabView, but the height of the built-in matrix buttons.
For example, in the PC emulator (800x480), the tabView's selection bar height is the same as the selection bar height I used on the small screen.

[image]

I need to adjust the height of this section, but I can't find the relevant place in the document

embeddedt, 2020-07-02:
You can probably change the padding of `LV_TABVIEW_PART_TAB_BG`.

kisvegabor, 2020-07-02:
> Sorry, I was too hasty. I'm not talking about the height of the tabView, but the height of the built-in matrix buttons.
For example, in the PC emulator (800x480), the tabView's selection bar height is the same as the selection bar height I used on the small screen.

I understood it this way, and liked the relevant part of the docs:
> LV_TABVIEW_PART_TAB_BG: The tab buttons which is a Button matrix. Clicking on a button will scroll LV_TABVIEW_PART_BG_SCRL to the related tab's content. The tab buttons can be accessed via LV_TABVIEW_PART_TAB_BTN. The height of the tab's button matrix is calculated from the font height plus padding of the background's and the button's style.

LusterWong, 2020-07-02:
> I understood it this way, and liked the relevant part of the docs:

I have read this part, but I don't know how to change the height of this area. Could you provide an example?Thank you

pete-pjb, 2020-07-02:
Hi @LusterWong,

I think to achieve a minimal height you would need to do something like this:

```c
	lv_obj_set_style_local_pad_top(tv, LV_TABVIEW_PART_TAB_BG, LV_STATE_DEFAULT, 1);
	lv_obj_set_style_local_pad_bottom(tv, LV_TABVIEW_PART_TAB_BG, LV_STATE_DEFAULT, 1);
	lv_obj_set_style_local_pad_top(tv, LV_TABVIEW_PART_TAB_BTN, LV_STATE_DEFAULT, 1);
	lv_obj_set_style_local_pad_bottom(tv, LV_TABVIEW_PART_TAB_BTN, LV_STATE_DEFAULT, 1);
```

Kind Regards,

Pete

LusterWong, 2020-07-03:
> Hi @LusterWong,
>
> I think to achieve a minimal height you would need to do something like this:
>
> ```c
> 	lv_obj_set_style_local_pad_top(tv, LV_TABVIEW_PART_TAB_BG, LV_STATE_DEFAULT, 1); [...]

Thank you so much!It does reduce the height!
But I still don't understand why. Haha im stupid.

kisvegabor, 2020-07-03:
Not stupid, you can't expect to know things without learning them. :slightly_smiling_face: 

Here is the description of padding: https://docs.lvgl.io/latest/en/html/overview/style.html#padding-and-margin-properties

pete-pjb, 2020-07-03:
Hi @LusterWong, I'll close this issue now, please feel free to re-open it if you require further assistance 😄

closed 2020-07-03 by pete-pjb