lvgl/lvgl #1639: lv_async_task_cb access violation in lvgl v6

Jamesits, 2020-07-05:
**Describe the bug**
On ESP32 (ESP-IDF 3.3), lvgl occasionally throws LoadProhibited exception at address 0x00000004.

The problem happen in `static void lv_async_task_cb(lv_task_t *task)` where `info` is NULL causing `info->cb` to fail.

**To Reproduce**
Not sure how to reproduce, it happens randomly every few hours in our production code. 

**Expected behavior**
It should not throw this exception.

**Screenshots**

Exception: 
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d8656  PS      : 0x00060a30  A0      : 0x800d8d75  A1      : 0x3ffc6e70
A2      : 0x00000000  A3      : 0x3ffcc358  A4      : 0x3ffcc30c  A5      : 0x3ffcc308  
A6      : 0x00000032  A7      : 0x00060023  A8      : 0x3ffcc304  A9      : 0x3ffcc300
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffd554c  A13     : 0x00000001  
A14     : 0x0000cdcd  A15     : 0x00000000  SAR     : 0x00000001  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000
```

Backtrace:
```
0x400d867a: lv_async_task_cb at lv_async.c:?
0x400d8d96: lv_task_exec at lv_task.c:?
0x400d8e2f: lv_task_handler at ??:?
0x400d3967: __lvgl_wrapper_draw_task at lvgl_wrapper.c:?
0x4008cf51: vPortTaskWrapper at port.c:?
```

**Additional context**

Simply return if `info` is NULL seems to make everything work for now:

```
lv_async_info_t *info = (lv_async_info_t *)task->user_data;
if (info == NULL) return;
```

I've not tested lvgl v7 since it is not possible to upgrade our code base for now.

embeddedt, 2020-07-05:
It sounds like your code is experiencing memory corruption. `lv_async_call` [checks that `info` is not `NULL`](https://github.com/lvgl/lvgl/blob/9e56f750a7f424d0cd52ce5ba3d4ccfa82a97866/src/lv_misc/lv_async.c#L41-L44). Something must be changing that value.

Jamesits, 2020-07-06:
Thank @embeddedt for the info. I've run some memory tests and showed no hardware issues. Don't have JTAG at hand so I tried printing logs everywhere. That's what I got:

```
I (1596909) LV: lv_async_call info == 0x3FFCD2AC
I (1596933) LV: lv_task_exec info == 0x3FFCD2AC
I (1596935) LV: lv_async_call info == 0x3FFCD2C4
I (1596995) LV: lv_task_exec info == 0x3120574F
```

This suggests occasionally `task->user_data` was modified somewhere between `lv_async_call` and `lv_task_exec`. I'm pretty sure nobody explicitly modified that pointer so I'm working on some more memory debugging.

kisvegabor, 2020-07-06:
@Jamesits Can you reproduce it with a really simple code to be sure it's the issue of `async`?
Or disable some parts of your code to narrow to potential places of the issue?

BTW, what does "occasionally" means? In every few minutes or every day?

Jamesits, 2020-07-06:
After further digging I think I can draw the conclusion that sometimes `lv_task_exec` is called from a second CPU core which resulted in a race condition, causing `lv_task_exec` mixing up 2 tasks from 2 `lv_async_call`s. I've made all lvgl function calls sequential and pinned it to a single core and it runs fine since then.

Given lvgl's nature of not supporting multithreading I think this is a non-issue. Closing.

closed 2020-07-06 by Jamesits