lvgl/lvgl #1645: run this project

mostafajalal20, 2020-07-08:
how can i run this project on vscode please guide me because i dont want to use other ide.

kisvegabor, 2020-07-08:
Which OS do you use?

mostafajalal20, 2020-07-08:
i have macOs and linux i prefer linux but if you can guide me for both of them I will thank you.

mostafajalal20, 2020-07-08:
please tell me how i run this project (lvgl-master) with vscode because i should add persian for writing on this project.

embeddedt, 2020-07-08:
Two options:

* You could use the [Eclipse project](https://github.com/lvgl/lv_sim_eclipse_sdl) and open it with VSCode instead. Then you can use the Makefile in the root directory to compile the project.
* You could use [`lv_platformio`](https://github.com/lvgl/lv_platformio) with PlatformIO (which is a VSCode extension).

I'm closing this as it is a question/support request, not a feature or development discussion. Please use the [forum](https://forum.littlevgl.com/) to ask questions.

closed 2020-07-08 by embeddedt