lvgl/lvgl #1647: Chart in circular mode doesn't draw below zero

DaPa, 2020-07-08:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

Have chart with data series in range 0 to 25 and chart points radius = 5. In "Shift" update mode the zero points are drawn properly (centred to zero). In "Circular" update mode they are cut to zero and any old drawings are not cleaned (please see recording). 

**To Reproduce**

Attached the example project.

**Expected behavior**

In circular mode the same area is drawn as in shift mode (a bit subzero if points radius > 0 and points gets close to zero).

**Screenshots**
Capture attached.

**Additional context**

LVGL 7.1.0 on Visual Studio simulator.
[image]
[chart_test.zip](https://github.com/lvgl/lvgl/files/4892900/chart_test.zip)

kisvegabor, 2020-07-09:
Hi,

Thanks for the report!

I've fixed it here: 416ef9e2

Can you confirm if it works well?

DaPa, 2020-07-09:
[image]
Perfect! :-)

closed 2020-07-09 by DaPa

kisvegabor, 2020-07-09:
Thanks for the feedback! :slightly_smiling_face: