lvgl/lvgl #1648: issue for setting critical value of gauge

ZhaoQiang-b45475, 2020-07-09:
Hi, 
I meet an issue, when setting critical value to a value between two adjacent labels, the arc and the scale color are unpredictable, please refer to the images attached, I set the critical as 124, 126, 180 and 121. is it a issue.

[image]
[image]
[image]
[image]

kisvegabor, 2020-07-10:
Hi,

I pushed a fix to master. Could you test it?

ZhaoQiang-b45475, 2020-07-14:
Thank you! I will try

kisvegabor, 2020-08-03:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2020-08-03 by kisvegabor