lvgl/lvgl #1649: lv_canvas_fill_bg dont work for LV_OPA_TRANSP??

incity, 2020-07-09:
lv_canvas_fill_bg(canvas, LV_COLOR_SILVER, LV_OPA_TRANSP);
fill the canvas with LV_COLOR_SILVER, and set transparency. but doesn't work for transparency.

kisvegabor, 2020-07-10:
Please send a code to reproduce.

incity, 2020-07-15:
lv_draw_rect_dsc_t rect_dsc;
    lv_draw_rect_dsc_init(&rect_dsc);
    rect_dsc.radius = 10;
    rect_dsc.bg_opa = LV_OPA_COVER;
    rect_dsc.bg_grad_dir = LV_GRAD_DIR_HOR;
    rect_dsc.bg_color = LV_COLOR_RED;
    rect_dsc.bg_grad_color = LV_COLOR_BLUE;
    rect_dsc.border_width = 2;
    rect_dsc.border_opa = LV_OPA_90;
    rect_dsc.border_color = LV_COLOR_WHITE;
    rect_dsc.shadow_width = 5;
    rect_dsc.shadow_ofs_x = 5;
    rect_dsc.shadow_ofs_y = 5;

    lv_draw_label_dsc_t label_dsc;
    lv_draw_label_dsc_init(&label_dsc);
    label_dsc.color = LV_COLOR_YELLOW;

    static lv_color_t cbuf[LV_CANVAS_BUF_SIZE_TRUE_COLOR(CANVAS_WIDTH, CANVAS_HEIGHT)];

    lv_obj_t * canvas = lv_canvas_create(lv_scr_act(), NULL);
    lv_canvas_set_buffer(canvas, cbuf, CANVAS_WIDTH, CANVAS_HEIGHT, LV_IMG_CF_TRUE_COLOR);
    lv_obj_align(canvas, NULL, LV_ALIGN_CENTER, 0, 0);
    lv_canvas_fill_bg(canvas, LV_COLOR_SILVER, LV_OPA_TRANSP);

    lv_canvas_draw_rect(canvas, 70, 60, 100, 70, &rect_dsc);

incity, 2020-07-15:
@kisvegabor Doesn't canvas support transparency?

pete-pjb, 2020-07-15:
Hi @incity,

I believe you need to set your canvas buffer image type to LV_IMG_CF_TRUE_COLOR_ALPHA in order to support transparency, LV_IMG_CF_TRUE_COLOR doesn't have the alpha channel required to support transparency.

I hope that helps.

Kind Regards,

Pete

incity, 2020-07-15:
@pete-pjb oh ok, thank you

closed 2020-07-15 by incity