lvgl/lvgl #1650: Gesture detection doesn't work on ESP32 based TTGO-T-Watch-2020

AlexGoodyear, 2020-07-09:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

The code in your example https://forum.lvgl.io/t/how-to-create-a-swipe-left-right-event/2379 does not work. The gesture events never occur.

**To Reproduce**

Use your own code on an ESP32.

**Expected behavior**

I expect the gesture event to be detected along with a direction

**Screenshots**
N/A

**Additional context**

I enabled logging and wrote a serial output callback and traced the problem to indev_gesture() in lv_core/lv_indev.c. I had to change from ...
    while(gesture_obj && lv_obj_get_gesture_parent(gesture_obj)) {
        gesture_obj = lv_obj_get_parent(gesture_obj);
    }
... to ...
    while(gesture_obj && lv_obj_get_gesture_parent(gesture_obj) && lv_obj_get_parent(gesture_obj)) {
        gesture_obj = lv_obj_get_parent(gesture_obj);
    }

It looks to me like the code was always going off the end of the parent list so gesture_obj was always NULL and would exit on the lines following this while() loop.

kisvegabor, 2020-07-10:
Please send a simple code snippet to reproduce.

embeddedt, 2020-07-31:
@kisvegabor It seems that it was [your code snippet on the forum](https://forum.lvgl.io/t/how-to-create-a-swipe-left-right-event/2379/2?u=embeddedt) which was used to reproduce this.

kisvegabor, 2020-08-03:
I see, thanks!

By default screen has  [gesture_parent = 0](https://github.com/lvgl/lvgl/blob/c85295b5ea9e44a5da67b12b8a7a9db179580360/src/lv_core/lv_obj.c#L336) so the loop should stop becasue `lv_obj_get_gesture_parent(gesture_obj) == false` for the screens.

Manually enabling  `gesture_parent` for the screen can cause the issue you described. Was this the case?

AlexGoodyear, 2020-08-03:
I checked my code and I do not have any calls to lv_obj_set_gesture_parent().

I do have a container declared as the first child of the default screen. If I attach an event handler to this child it will not detect gesture events unless I have an lv_obj_set_gesture(container, 0) call.

I accept that using the lv_obj_set_gesture_parent(object, 0) is preferential to changing the core library code.

When I submitted this issue notice, the library I was using (Xinyuan-LilyGO/TTGO_TWatch_Library) included version 6.x of LVGL but it has recently been upgraded to 7.x. I'm certain I was using the default screen in version 6.x as your example but that doesn't seem to deliver any events (press, focus or gesture) to the callback in version 7.x

kisvegabor, 2020-08-04:
> I do have a container declared as the first child of the default screen. If I attach an event handler to this child it will not detect gesture events unless I have an lv_obj_set_gesture(container, 0) call.

You should see the gesture event in the screens event handler because gesure_parent is enabled for every non-screen object by default. 

Try this:

```c
void scr_event_cb(lv_obj_t * obj, lv_event_t e)
{
    if(e == LV_EVENT_GESTURE) {
        lv_gesture_dir_t dir = lv_indev_get_gesture_dir(lv_indev_get_act());
        printf("Dir: %d\n", dir);
    }
}

 ...

  lv_ex_btn_1();
  lv_obj_set_event_cb(lv_scr_act(), scr_event_cb);
```

AlexGoodyear, 2020-08-04:
I do not have a function lv_ex_btn_l() [ell] or lv_ex_btn_1() [one] in my version of lvgl ... from the CHANGE.log file ...
# Changelog
## v7.0.0 (18.05.2020)

embeddedt, 2020-08-04:
It's part of `lv_examples`.

AlexGoodyear, 2020-08-04:
Looks like the examples directory has been removed from the embedded version of lvgl in the library I've included

embeddedt, 2020-08-04:
In that case, you can just copy and paste [the relevant functions](https://github.com/lvgl/lv_examples/blob/659ffeb82f7d9402b02a4182f3dd6e3fa852bce1/src/lv_ex_widgets/lv_ex_btn/lv_ex_btn_1.c#L5-L35) from `lv_ex_btn_1.c`.

kisvegabor, 2020-08-04:
It just creates 2 buttons, you can do it yourself as well.   I've used the button example to have some elements to test with.

AlexGoodyear, 2020-08-04:
Ok, I found the equivalent example in the [TTGO-Twatch library examples](https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/blob/master/examples/LVGL/Lvgl_Button/Lvgl_Button.ino)

I'll add the event handler code for the lv_scr_act() to this and test ...

OK, I only get the Toggled and Clicked messages, no gesture events when I swipe.

If I create a container and then attach the event_cb to that and execute an lv_obj_set_gesture_parent(cont,0) I get the the gesture dir events printed. If I remove the lv_obj_set_gesture_parent() I don't get any gesture dir events. I double checked these results

wizche, 2020-08-04:
Same here, no gesture events are being received

kisvegabor, 2020-08-06:
Please try this:
```c
  printf("screen gesture parent: %d\n", lv_obj_get_gesture_parent(lv_scr_act()));
```

I got:
```
screen gesture parent: 0
```

AlexGoodyear, 2020-08-06:
I got ...

screen gesture scr_act: 1
screen gesture cont: 1

After doing ... lv_obj_set_gesture_parent(cont,0);

screen gesture scr_act: 1
screen gesture cont: 0

kisvegabor, 2020-08-06:
Do you use the latest version of lvgl?

AlexGoodyear, 2020-08-06:
I use a version embedded into the [TTGO-Twatch library](https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library)
which is v7.0.0 (18.05.2020)

kisvegabor, 2020-08-06:
In that version of LVGL it still looked different: https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/blob/master/src/lvgl/src/lv_core/lv_obj.c#L313

So it set `gesture_parent` to screens too. You can either update lvgl or use the workaround you have found (manually setting gesture parent of the screen to zero)

AlexGoodyear, 2020-08-06:
OK, thank you.

closed 2020-08-06 by AlexGoodyear