lvgl/lvgl #1651: How to change the order of focus(LV_INDEV_TYPE_ENCODER)?

LusterWong, 2020-07-10:
My system has three keys, so I use encoder input mode(LV_INDEV_TYPE_ENCODER).

I set up a container with 2 switches and 4 buttons underneath, as shown below:
[image]
With the initial position at AAA, press the Next button to switch to the next object.The order I want is AAA-BBB-CCC-DDD-EEE-FFF.
But in a different order, is AAA-DDD-BBB-CCC-EEE-FFF.

Is there any way to change the default order?

kisvegabor, 2020-07-10:
Hi,

The order determined by in which order you add the object to the group. 

To change the order you can remove all objects from the group and add them again in the new order.

LusterWong, 2020-07-11:
> Hi,
> 
> The order determined by in which order you add the object to the group.
> 
> To change the order you can remove all objects from the group and add them again in the new order.
I made it. Thank you very much.

closed 2020-07-11 by LusterWong