lvgl/lvgl #1654: lv_img_cache broken design?

fpw, 2020-07-12:
**Describe the bug**

I investigated a very rare crash bug in my application and nailed it down to this:

- I'm opening images at runtime and assign them to an `lv_img` by creating an `lv_img_dsc_t` on the heap and using that with `lv_img_set_src`.
- The `data` pointer in my `lv_img_dsc_t` is created by myself and also a pointer to the heap, i.e. I'm decoding the data for the image myself
- There seems to be a cache in LVGL that maps from my `lv_img_dsc_t` heap pointers to `lv_img_cache_entry_t` pointers based on the pointer address.

Now, what causes the crash is this:
- I'm deleting an `lv_img` using `lv_obj_del`
- After that, I also free the memory block in the `data` pointer of `lv_img_dsc_t`
- I'm creating a new image with the steps described above
- **Due to random circumstances, the heap allocations returns an address that was also used by a previous image thas has already been deleted!**
- Now the cache returns the cached `lv_img_cache_entry_t` that contains a deleted pointer for the image data!
- This results in a crash

As a workaround, I'm now calling `lv_img_cache_invalidate_src` before passing my heap pointer into `lv_img_set_source`.

**To Reproduce**

This is not easily reproduced because it depends on the addresses returned by `malloc`. I hope the steps explained above are sufficient to understand the root cause.

**Expected behavior**

LVGL should not assume that `src` pointers passed into `lv_img_set_src` are unique. This is not the case when using the heap. 

**Screenshots**

None

**Additional context**

I'm not sure if the current design allows to fix this because LVGL does not know when a `data` pointer becomes invalid.
Maybe there should be a way to tell LVGL that the image was already decoded and the cache is not requried?

I did this on LVGL 6 but I think LVGL 7 is also affected because the design didn't seem to have changed.

kisvegabor, 2020-07-13:
Hi,

Thanks for the clear description of the issue. 

Actually `lv_img_cache_invalidate_src` is created exactly for this purpose. Sorry if this caused headache for you but I can't imagine a better way to handle this now. :slightly_frowning_face: 

Automatically calling `lv_img_cache_invalidate_src` on deletion of `lv_img` would be possible but caused unesseray cache invalidations.

fpw, 2020-07-13:
Alright, I understand 🙂 Thanks for the clarification, I'll live with that then.

closed 2020-07-13 by fpw