lvgl/lvgl #1655: LV_EVENT_CANCEL not handled by list/group

tehkillerbee, 2020-07-13:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

In our design, we are using a keypad to navigate different objects, including a list. The keypad has RIGHT/LEFT/UP/DOWN/ENTER and CANCEL. In this specific case, I have mapped the CANCEL key to ESC, as there is no specific CANCEL key in the lv_key_t.  However, the CANCEL does nothing when pressed. The list object simply does not handle this event, whenever the ESC / BACKSPACE key is pressed. However, the btn.

**To Reproduce**

Create a list object, assign group to it to make it possible to navigate using keypad. 
The keypad should have RIGHT/LEFT/UP/DOWN/ENTER and CANCEL keys. CANCEL is mapped to ESC.

**Expected behavior**

All keys except CANCEL works in the above situation

**Screenshots**
NA

**Additional context**
As far as I can see, lv_dropdown is the only widget that handles ESC directly.  However, I needed this functionality for lv_list so I added it:
https://github.com/lvgl/lvgl/blob/master/src/lv_widgets/lv_list.c#L769
`        
...
else if(c == LV_KEY_LEFT || c == LV_KEY_UP) {
            lv_list_ext_t * ext = lv_obj_get_ext_attr(list);
            /*If there is a valid selected button the make the next selected*/
            if(ext->act_sel_btn != NULL) {
                lv_obj_t * btn_next = lv_list_get_prev_btn(list, ext->act_sel_btn);
                if(btn_next) lv_list_focus_btn(list, btn_next);
            }
            /*If there is no selected button the make the first selected*/
            else {
                lv_obj_t * btn = lv_list_get_next_btn(list, NULL);
                if(btn) lv_list_focus_btn(list, btn);
            }
        }
        else if(c == LV_KEY_ESC) {
            lv_list_ext_t * ext = lv_obj_get_ext_attr(list);
            /* Handle ESC/Cancel event */
            res = lv_event_send(ext->act_sel_btn, LV_EVENT_CANCEL, NULL);
        }`

kisvegabor, 2020-07-13:
Hi,

There is a trick with `lv_list`: although the list is focused the events are forwarded to the selected button. See here: https://github.com/lvgl/lvgl/blob/master/src/lv_widgets/lv_list.c#L667-L698

Your fix seems good but it'd be better to handle this in one place. Could you test it and send a Pull request?

tehkillerbee, 2020-07-13:
Thank you for your idea! Looks like that's the same method I used, when I needed to capture the ESC button on any other btn widget. I will see if I can get it working with a list item as well.

tehkillerbee, 2020-07-14:
After looking over my code again, it turns out I am using the callback for the list button itself to handle the event - eg. when a list button is pressed - just as I did with the "normal" button widgets.
However, no event never gets triggered when pressing CANCEL on a selected list button, hence the need for the above fix. When pressing ENTER, the event is forwarded to the selected button as usual, as you said. I guess the CANCEL event is not forwarded, simply because the CANCEL button is classed as a LV_SIGNAL_CONTROL type. 

The above fixes the issue - I will send a pull request ASAP.

tehkillerbee, 2020-07-14:
I looked further into this - and I figured out whats going wrong. The LV_EVENT_CANCEL event is handled - but it is not passed on to the button but only the list itself. The button event callback is never triggered, unless a button (ENTER) is pressed.

Now I am using separate event callback for both the list object and the list button object. This works as expected.
```
lv_obj_set_event_cb(this->s0.list_menu, list_event_handler);
lv_obj_set_event_cb(list_btn, list_btn_event_handler);
```
```
void GUI::list_event_handler(lv_obj_t * obj, lv_event_t event){
	/*get pointer to the list_state variable associated with the GUI class*/
	if(event == LV_EVENT_CANCEL) {
		//Handle CANCEL event in main GUI thread
                ...
        }
    return;
}
```
```
list_btn_event_handler(lv_obj_t * obj, lv_event_t event){
	/*get pointer to the prompt_state variable associated with the GUI class*/
	int8_t * list_btn_state_ptr = (int8_t *)lv_obj_get_user_data(obj);
	int32_t list_btn_idx = lv_list_get_btn_index(NULL,obj);

    if(event == LV_EVENT_CLICKED) {
    	/*get index of the list button pressed */
    	if(list_btn_idx>=0){
    		*list_btn_state_ptr = (int8_t)list_btn_idx;
    	}else{
    		*list_btn_state_ptr = -1;
    	}
    }
    return;
}
```

closed 2020-07-14 by tehkillerbee

kisvegabor, 2020-07-14:
That's also a good way to manage it. 

Anyway, I think sending the event to the button too is more intuitive.