lvgl/lvgl #1656: suggestion about lv_label_align_t

incity, 2020-07-13:
1. suggest to add LV_LABEL_ALIGN_JUSTIFY
2. suggest to remove LV_LABEL_ALIGN_AUTO, and do like this in Android:
[image]

[https://developer.android.com/reference/kotlin/androidx/ui/text/style/TextAlign](url)

kisvegabor, 2020-07-13:
Replacing AUTO with START/END sounds reasonable. It aligns well with the termes we are planning to use with the [new layouts](https://github.com/lvgl/lvgl/issues/1615#issuecomment-654981582).

JUSTIFY is also possible.

kisvegabor, 2020-08-04:
There will be big updates for v8 and it can be adding in it.

kisvegabor, 2020-08-31:
Finally, I added this as an idea to the [ROADMAP](https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md).

Thanks for the suggestion.

closed 2020-08-31 by kisvegabor