lvgl/lvgl #1657: [suggestion] Roadmap

puzrin, 2020-07-13:
Created after internal discussion.

LVGL passes series of big changes. With limited human resources, this can take some time. To minimize different kinds of "growth problems", it would be useful for all to understand "where everything goes" and "what happens next". Since tracker milestones may be not convenient at this moment, it would be nice to see roadmap in any free format (google doc, gist, ...).

I've prepeared a small draft how things could start https://gist.github.com/puzrin/12bf3095f44f902a5ac23bebcef4c592. It's not to duplicate every small issue from tracker. But it would be nice to understand state of "big decisions".

It's not necessary to make roadmap a "formal plan to follow". Having "intents at current moment" is enough (and much better than nothing).

kisvegabor, 2020-07-13:
Sounds reasonable. The  "intents at the current moment" approach seem doable. I wouldn't dare to add more serious statements. :slightly_smiling_face: 

So far we haven't planed more than one major release but maybe if write it down will see things better.

kisvegabor, 2020-07-24:
I've created the roadmap file: https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md

@puzrin 
Next week I'll consider your ideas listed on the link above.

kisvegabor, 2020-08-03:
I've added most of the ideas you suggested.

I removed the PIO related one to avoid sticking to a specific IDE.

kisvegabor, 2020-08-04:
Slightly off-topic.
I've just run Coverity (as you suggested among the ideas). There were 59 defects. Some of them are really serious.
[image]

puzrin, 2020-08-04:
Yeah, especially at first pass, linter [code analyzer] & coverage reports help to find a lot of unobvious defects. Code analyzer may be noisy, that's normal.

kisvegabor, 2020-08-10:
As the ROADMAP file is added and used I close this issue.

Thank you for the idea!

closed 2020-08-10 by kisvegabor