lvgl/lvgl #1658: [suggestion] Unit testing

puzrin, 2020-07-13:
Created after internal discussion.

While each of us has own opinion about importance of unit testing, i think everyone will agree, this will happen sometime. For example, i have no experience how to organize unit testing for C projects, but i would be happy to use it. It would be nice to understand priority of this feature, and what can be done to get it sooner.

- Last state of my understanding of technical details was posted here https://github.com/lvgl/lvgl/issues/1305#issuecomment-574442452
- My personal experience includes only using built-ins of PlatformIO.

kisvegabor, 2020-07-13:
Thanks for opening this issue. 

Honestly, I have limited experience with testing... In one of my former workplaces, I wrote unit tests but it looked like writing tests just for the sake of having tests. So it wasn't convinced me but I know (unit) tests are considered important.

But, do we really need UNIT tests? Focusing on coverage, testing a single file and stubbing others? What about functional tests? E.g. call `lv_obj_set_x(obj, 10)` and see if `obj->coords.x1 == 10` or `lv_obj_get_height(obj) == 10`.

I suppose both are important and one can't replace the other. However which one should have higher priority?

puzrin, 2020-07-13:
> But, do we really need UNIT tests? Focusing on coverage, testing a single file and stubbing others?

Yes. There are places where unit tests are mandatory (and simple). Remember, you had problems in decompressor implementation. Having tests like in JS could help a lot. And, no stubbing needed (and no real hardware).

I do not mean, everyone must be enforced to write unit tests. But everyone should be able to write those with easy, if desired.

> What about functional tests? E.g. call `lv_obj_set_x(obj, 10)` and see if `obj->coords.x1 == 10` or `lv_obj_get_height(obj) == 10`.

Probably here is confusion about terminology. I don't see principal difference in this things. There is another difference:

1. You can make unit tests - checking single file.
2. You can make integration tests - built whole system and, try to do something with it & see reaction.

Second approach may look more attractive at start, but i'm speaking about first one (in scope of this issue).

> Honestly, I have limited experience with testing...

I have no big experience of unit testing in C/CPP (only via PlatformIO). I have experience in JS & general understanding of approaches.

If you are ok about cmake/ctest/..., i think it's possible to copy-paste details from some high-quality project. I can not suggest something concrete/final, because have no eхperience in this area. May be https://github.com/ETLCPP/etl, but that's only my personal and very limited opinion.

kisvegabor, 2020-07-14:
> Second approach may look more attractive at start,

I agree.

I've found [CUnit](https://sourceforge.net/projects/cunit/) easier to use but haven't immersed in it deeply yet.

So there are some units (memory manager, linked list, tasks,  colors etc) which can be easily unit tested, but higher-level things (the new layout, input device handling, etc)  should be integration tested. Does it make sens?

puzrin, 2020-07-14:
> So there are some units (memory manager, linked list, tasks, colors etc) which can be easily unit tested, but higher-level things (the new layout, input device handling, etc) should be integration tested. Does it make sens?

- If you test input device handling with full lvgl assembly and tracking via fake output device - that will be interation test. Because lvgl acts as monolytic black box.
- But if you built only input device file (with mocks), and check it's output events or function call - that will be unit test.

Another example - decompressor. I can compare display images for compressed and non-compressed fonts (integration test). Or i can compare input and output data vectors (unit test).

Everything can be unit-tested, if you have enough mocks to replace real system :). The only question - how much do you need that and how difficult is it.

I can advice only one thing - don't try to replace unit tests with integration tests "because it's more easy". That's a common mistake for novices. Just do tests when you really wish and understand why, and everything will be fine.

kisvegabor, 2020-07-16:
Sound fine. 

I assume the next step is to set up a test environment.

C47D, 2020-07-18:
I've used Unity and fff (for mocking) at work to test an AT command parser, I've also seen a bit with CppTest, which as far as I know supports C and C++.

kisvegabor, 2020-07-21:
Unity looks great! It's only a C and H file and seems very easy to use and understand.

ValentiWorkLearning, 2020-07-23:
I've experience with using GTest and GMocka. Sure, they are suitable for your unit-tests goals.

puzrin, 2020-07-23:
@kisvegabor i'm for gtest+gmock because this fit into "independent poll results", referred in first post. If @ValentiWorkLearning can help with tuning this for lvgl, this would be awesome.

What do you think?

kisvegabor, 2020-07-24:
I'm ok with gtesr+gmock too.

kisvegabor, 2020-08-17:
It's added to the ROADMAP. 

Now I'm dealing with the features planned for v8 but if anybody has the interest to dive into this now, feel free to comment here. 

I close this issue until then.

closed 2020-08-17 by kisvegabor