lvgl/lvgl #1659: lv_event_send()missing an graphical synchronous change for an object?

LusterWong, 2020-07-13:
First I set up a switch object:
```
sw = lv_switch_create(Parent, NULL);
lv_obj_set_event_cb(sw, sw_cb);

void sw_cb(lv_obj_t *Obj, lv_event_t E)
{
	if(E == LV_EVENT_CLICKED)
	{
		printf("do something\n")
	}
}
```
When sw is pressed or clicked, it enters printf("do something\n").

But when I used LV_event_send (sw, LV_EVENT_CLICKED, NULL) to actively trigger, even though I could go into printf("do something\n"), the graph of sw didn't change as if I hadn't pressed it.

Did I have a problem with the Settings or did LVGL design this on purpose?

embeddedt, 2020-07-13:
Sending an event to the object won't trigger a click on the object. Events are designed for the object to notify you, not the other way around.

You should use the appropriate switch APIs to change the switch's value.

kisvegabor, 2020-07-16:
I close it as the question is answered.

closed 2020-07-16 by kisvegabor