lvgl/lvgl #1662: LV_FONT_SIMSUN_16_CJK doesn't work well

ZhaoQiang-b45475, 2020-07-15:
Hi I use LV_FONT_SIMSUN_16_CJK to set a label, and it doesn't display chinese charaters.
My code is as below:

**void test(){

    /* Create a style for the shadow*/
    static lv_style_t label_style;
    lv_style_init(&label_style);
	lv_style_set_text_font(&label_style, LV_STATE_DEFAULT, &lv_font_simsun_16_cjk);

    /* Create the main label */
    lv_obj_t * main_label = lv_label_create(lv_scr_act(), NULL);
	lv_obj_add_style(main_label, LV_STATE_DEFAULT, &label_style);
    lv_label_set_text(main_label, "\xE9\x9A\xA8" "\xE9\x87\x8F" "\xE9\x87\x9D" "a");

}**
"\xE9\x9A\xA8" "\xE9\x87\x8F" "\xE9\x87\x9D" are "隨量針"(96A8,91CF,91DD), the result as below:
[image]

kisvegabor, 2020-07-15:
Hi,

I saw this issue in an earlier version. Please, update to the latest version from `master` and try again.

ZhaoQiang-b45475, 2020-07-16:
yes, part of them work with new version, but some others still doesn't work, like "\xE4\xB8\x80" "\xE4\xB8\x83" ("一" "七")

kisvegabor, 2020-07-16:
It works well for me:
```c
lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
lv_obj_set_style_local_text_font(label, 0, 0, &lv_font_simsun_16_cjk);
lv_label_set_text(label, "一七");
```

closed 2020-08-09 by embeddedt