lvgl/lvgl #1665: Terms update, style => class?

puzrin, 2020-07-15:
In lvgl term "style" has historic reasons, but with new api seems confusing. Because in CSS/HTML, "style" is "content of class". Now we assign classes to objects (on scope of CSS/HTML).

Is it realistic to update api methods names & doc? Not nice breaking change, but can be compensated in long term.

embeddedt, 2020-07-15:
We could change the API names if macros were left in place to ensure backwards compatibility.

kisvegabor, 2020-07-15:
As the analogy between CSS-LVGL is not obvious at first look we can't rely on
- people know the terms might come from CSS
- we can't assume users know CSS

Do you think "class" is a more intuitive name?

Personally I think the term "style" is more descriptive. If I saw the use of class I asked: What is a class? Is it related to C++? Or class of what? Is it like a group of something? Add class to an object? Rather an object should be added to a class?

pete-pjb, 2020-07-15:
Hi, I agree with @kisvegabor here as a person coming from a pure C/Assembler background I know very little about HTML/CSS etc.

puzrin, 2020-07-15:
> Do you think "class" is a more intuitive name?

That's not critical, but "style" at current place cause some blood from my eyes :).

> Personally I think the term "style" is more descriptive.

IMO it's descriptive in context of your "knowledge about world at current moment". This may change.

> If I saw the use of class I asked: What is a class? Is it related to C++? Or class of what? Is it like a group of something? Add class to an object? Rather an object should be added to a class?

Sounds reasonable. Let me generalize problem to "Term `style` could be replaced with something more appropriate". `cssclass` and so on (only as example). I agree, term `class` will be too confusing for programmers. But IMO `style` is far from perfect too.

What about QT? This lib has "comparable" approach. Which terms does it use?

kisvegabor, 2020-07-16:
> That's not critical, but "style" at current place cause some blood from my eyes :).

I hope you get better soon. :smile: 

> What about QT? This lib has "comparable" approach. Which terms does it use? I would agree with any well known convention.

QT uses the term [StyleSheet](https://doc.qt.io/qt-5/stylesheet-examples.html). Their styling system is very CSS-like.

In GTK if the styles are described by CSS files they can be applied by [Gtk.StyleContext.add_provider_for_screen](https://gist.github.com/carlos-jenkins/8923124#file-app-py-L62). "Style" is also the [name of an object](https://developer.gnome.org/pygtk/stable/class-gtkstyle.html) describing the appearance of widgets

puzrin, 2020-07-16:
May be `stylesheet`...

Ok, let's close this until become more actual.

closed 2020-07-16 by puzrin