lvgl/lvgl #1666: Deleting chart object with one or more series results in heap double free

getairq, 2020-07-15:
> # Important: issues that don't use this template will be ignored/closed.

I malloc and create a chart when users press a button on the UI to view (I wanted to save RAM).  This worked fine in v6.x (now using LVGL 7.1.0), but under FreeRTOS trips a heap assert for freeing already free memory.   I also had LV_DEBUG enabled with all checks enabled (no LVGL fault triggered).
The fault only occurs once a series is added to the chart (I commented out code until finding the line that caused the problem). 
This fault does not occur using the Visual Studio PC simulator.  I am using the standard heap_4.c memory allocator on FreeRTOS v10.3.0
The problem happens both when an external array is used or internal array for the chart series.

**To Reproduce**

// Chart parent object:
    params->screen = lv_obj_create( NULL, NULL );
    lv_obj_add_style( params->screen, LV_OBJ_PART_MAIN, &style_screen );
    lv_obj_set_size( params->screen, hres, vres );
    lv_obj_t* wall_paper = lv_img_create( params->screen, NULL );
    lv_img_set_src( wall_paper, &hex_dark_3 );

// Chart creation (done on demand)
            local->history_chart = lv_chart_create( local->screen_info.screen, NULL );
            lv_chart_set_type( local->history_chart, LV_CHART_TYPE_LINE );
            lv_chart_set_point_count( local->history_chart, 13 );
            local->history_chart_series_min = lv_chart_add_series( local->history_chart, LV_COLOR_BLACK );

// Chart delete (done on exit)
            lv_obj_del( local->screen_info.screen );

**Expected behavior**

No run-time asserts.

**Screenshots**

No screen shots - this happens on my target hardware (a Cortex-M4 based custom PCB)

**Additional context**

Nil

kisvegabor, 2020-07-16:
It should be fixed here: https://github.com/lvgl/lvgl/commit/baeda1998fdecea9ce611816f058b5fe165ab451#diff-f606fafb06c643e20a51b03539159232

Can you confirm if it fixes the issue?

getairq, 2020-07-16:
Confirmed this fixes the issuse.

closed 2020-07-16 by getairq