lvgl/lvgl #1667: spinbox can no set the height with the function lv_obj_set_height

RanBiron, 2020-07-16:
>  lv_obj_t * spinbox = lv_spinbox_create(cont, nullptr);
  lv_obj_set_width(spinbox , LV_DPX(100));
  lv_obj_set_height(spinbox , LV_DPX(36));

I create a spinbox with the function lv_spinbox_create()，and I can set width with function lv_obj_set_width(),but I can no set height
with the function lv_obj_set_height().why?

embeddedt, 2020-07-16:
The spinbox inherits from the text area, which sets its height automatically based on the font in use.

kisvegabor, 2020-07-21:
As the question is solved I close this issue.

---

We use GitHub Issues for development-related discussions.

Please ask questions in the [Forum](https://forum.lvgl.io) to make it visible for a larger community. You can simply log in with your GitHub account.

closed 2020-07-21 by kisvegabor