lvgl/lvgl #1668: Color Picker (LV Cpicker) edge serrated especially badly.

Accidentaly-across, 2020-07-16:
Hi 

I found that the running effect of the Color Picker control has obvious jagged edges. Can this problem be optimized?

Accidentaly-across, 2020-07-16:
[image]

embeddedt, 2020-07-16:
It's a limitation of the way the color picker is rendered.  I thought it was no longer an issue in 7.0 since the edges are masked anyways, but this appears not to be the case.

closed 2020-07-24 by kisvegabor

pete-pjb, 2020-07-24:
Hi @Accidentaly-across,

I have improved the edges of the colour picker widget.  The changes have been added as a hot fix in the master branch of the LVGL repository.

Kind Regards,

Pete