lvgl/lvgl #1669: Code size increase in 7.0+

embeddedt, 2020-07-16:
In a discussion with the [Zephyr project](https://github.com/zephyrproject-rtos/zephyr/pull/26654) regarding them updating to 7.0+, it was mentioned that 7.0 has [significantly increased the Flash space requirements](https://github.com/zephyrproject-rtos/zephyr/pull/26654#pullrequestreview-446548076). In some cases the requirement has doubled or even tripled.

In the `size_reduction` branch, I've found and addressed a few config-related issues that resulted in more space being used than necessary, but I think there is still a lot more that can be done to reduce Flash requirements. Ideally, one should be able to turn off most/all of the new features in 7.0 (besides the style system) and get a code footprint similar to that of 6.1.

It appears that the `lv_draw` functions are the main culprit here: together they take up about 10K of space.

kisvegabor, 2020-07-17:
All changes look appropriate. Have measured its impact on code size?

I also believe the drawing functions can be more optimal regarding flash usage.  (I was focusing on speed rather than code size)

For an estimate how much can we win from 10k? Maybe 3k? It seems there are larges problems here if the code size was doubled.

embeddedt, 2020-07-17:
It appears to save a few kilobytes when the appropriate options are disabled. If they're enabled then it makes no difference.

In `lv_draw_rect.c` I noticed that `draw_border` and `draw_outline` look quite similar. Are they candidates for de-duplication?

Perhaps some of the logic pertaining to masks can be made configurable? That seems to be a what a lot of the code is dealing with.

Since there's already an option to disable the `opa_scale` property, maybe that should be broadened to disable support for opacity in the whole library? That might not work so well with antialiasing though.

kisvegabor, 2020-07-17:
> In lv_draw_rect.c I noticed that draw_border and draw_outline look quite similar. Are they candidates for de-duplication?

It seems correct. Some parts should common. I'll work on it next week.

> Perhaps some of the logic pertaining to masks can be made configurable? That seems to be a what a lot of the code is dealing with.

I've checked it and the 2 largest masks (radius and line) are "mandatory". In fact in the [report](https://gist.github.com/gmarull/ec8c3fbcf5dbcc26801363a03e9566ff#file-rom_report_v7_empty_unscii8-txt-L194-L204) only these 2 masks are added.  Probably there are some room for optimization here too.  

> Since there's already an option to disable the opa_scale property, maybe that should be broadened to disable support for opacity in the whole library? That might not work so well with antialiasing though.

Yes, it would break anti-aliasing. :slightly_frowning_face: `opa_scale` can be disabled to avoid a lot of inherited style querries usually for nothing.

embeddedt, 2020-07-17:
> Yes, it would break anti-aliasing.

Would that be a problem though if `LV_ANTIALIAS` is already disabled?

kisvegabor, 2020-07-17:
> Would that be a problem though if LV_ANTIALIAS is already disabled?

It requires a deeper examination to so all the possible issues. However, at first look no opacity could work with no anti-aliasing.  

We need to keep in mind to make optional only the really critical parts because too many options can make maintenance quite difficult.

C47D, 2020-07-18:
Hi @embeddedt, @kisvegabor,

I will check the Zephyr Kconfig files and compare them to the Kconfig on the lv_port_esp32 repo, maybe we can learn something from it or even add the Kconfig file directly on the LVGL repo as planned.

embeddedt, 2020-07-18:
@C47D I'm not sure what the license of the Zephyr Kconfig file is; it might be worth looking into that first.

vanwinkeljan, 2020-07-20:
@embeddedt zephyr files are apache2 licensed

@C47D make sure to pick the Kconfigs from the Zephyr PR mentioned above or you will end up with V6 Kconfig files

Further the zephyr Kconfig follows an approach to only enable a minimal amount of features by default so that the user can pick only the features required by his application.

kisvegabor, 2020-07-21:
I added `draw_full border` for the common parts of outline and border drawing and added the `LV_USE_FONT_SUBPX` option. It makes the code 2-3 kB smaller.

Do you know where are the settings which were used when the code size doubling was measured?

vanwinkeljan, 2020-07-22:
@kisvegabor just double checked and it looks like with v6, zephyr has theme support disabled and in the v7 PR the material theme is used. Switching to an empty theme with v7 reduces the rom usage with 4k but still there is a delta of 20k compared to v6.

A quick overview of v6 vs v7 (with empty theme):

```
lib                 22233    37.80%                           | lib                42049    51.72%
  gui               22233    37.80%                           |   gui              42049    51.72%
    lvgl            22233    37.80%                           |     lvgl           42049    51.72%
      src           22233    37.80%                           |       src          42049    51.72%
        lv_core      5636     9.58%                           |         lv_core     9760    12.00%
        lv_draw      7702    13.10%                           |         lv_draw    20484    25.20%
        lv_font      2609     4.44%                           |         lv_font     2791     3.43%
        lv_hal        482     0.82%                           |         lv_hal       554     0.68%
        lv_misc      3098     5.27%                           |         lv_misc     5340     6.57%
        lv_objx      2694     4.58%                           |         lv_widgets  2952     3.63%
        lv_themes      12     0.02%                           |         lv_themes    168     0.21%
===========================================                   | ==========================================
                    58812                                     |                    81300
```

Full reports can be found at https://gist.github.com/vanwinkeljan/607fa993c36558cdb4236d99e7da5f17

kisvegabor, 2020-07-24:
Thanks, @vanwinkeljan!

In v6, the line an image drawing were disabled when the line and image widgets were disabled. It's not the case in v7. It means ~4K extra ROM usage. 

The updates in `size_reduction` branch should reduce with an other ~5k. 

Could you check the current state with this branch?

`lv_draw_arc` and `lv_draw_triangle` are not present in v7 report. Might they be ignored during linker optimization?

BTW, what optimization level was used fr the reports? If O0 could you compare with Os too?

vanwinkeljan, 2020-07-24:
@kisvegabor tested with the `size_reduction` branch and there was a reduction of 2,7k with `LV_USE_FONT_COMPRESSED` and `LV_USE_FONT_SUBPX` disabled.

Full log:
https://gist.github.com/vanwinkeljan/607fa993c36558cdb4236d99e7da5f17#file-lvgl-v7-rom-report-size_reduction-eb725a3a

Further zephyr is always building with Os by default

kisvegabor, 2020-07-24:
Thank you. 

It's not that much... :slightly_frowning_face: 

Most of the difference comes from `lv_draw` (+10k). By excluding `lv_draw_line` and `lv_draw_img` we can win ~4k.
And with some small optimization and others maybe an other ~5k.
Not a big gain. And the new (better) features have a cost. That is we can't expect to keep the same size while adding features.

So maybe we can win a few kB here and there, but to really make a difference we should reconsider a few things. Something like making the core functionalities very lean and push some features to the user to handle in events or so.

Besides with [Simian](https://www.harukizaemon.com/simian/index.html) I measured that there are about 1000 duplicated lines in lv_core+lv_draw+lv_msic and other 1000 in lv_widgets.  Looking into the could also help reduce with a few kB.

puzrin, 2020-07-25:
With my kind of projects, at 6.1 i have ~ 10-15K of reserve, when development still stays comfortable (i don't like to drop printf & retargeting to USB, floats and so on).

My minimal checkpoint means STM32F0 with 128K flash, and 3-4 subpiхel fonts for android-like high quality interface.

I'd suggest to land quick-fixes and wait with the rest until new breaking changes complete (layout, scrolling, true RGBA for styles, ...). This should help with codebase cleanup and may change optimization priorities too much.

---

I could suggest fonts packing into "catalogs" to reuse kerning tables, but it's too early IMHO.

kisvegabor, 2020-07-27:
> My minimal checkpoint means STM32F0 with 128K flash

I definitely agree to ensure LVGL can work on such a device. 

> I'd suggest to land quick-fixes and wait with the rest until new breaking changes complete (layout, scrolling, true RGBA for styles, ...). This should help with codebase cleanup and may change optimization priorities too much.

I also agree to add it in v8.

@embeddedt @pete-pjb 
Do you also agree to make a big API clean up in v8?

> I could suggest fonts packing into "catalogs" to reuse kerning tables, but it's too early IMHO.

`fonts` is a well-separated module and we can treat them independently from the core. 

---

In #[1675](https://github.com/lvgl/lvgl/issues/1675#issuecomment-663872276) performance issues were reported and the proposed solution was to make possible disabling some rarely used features. 

It will reduce the size further with another few kB.

kisvegabor, 2020-07-27:
@embeddedt do you agree to merge `size reduction` branch.

embeddedt, 2020-07-27:
> Do you also agree to make a big API clean up in v8?

I agree, although with the usual recommendation of macros to prevent unnecessary work finding+replacing.

> do you agree to merge the `size_reduction` branch into `master`.

Yes; sounds good.

pete-pjb, 2020-07-27:
Hi @kisvegabor ,

> Do you also agree to make a big API clean up in v8?
Yes I think that is a good plan, is there a plan of action for it?  I could possibly help out with some of it if there are some tasks listed. (Although I am on holiday for two weeks from this coming weekend so will probably be off line for most of that period) 

Kind Regards,

Pete

kisvegabor, 2020-07-27:
> I agree, although with the usual recommendation of macros to prevent unnecessary work finding+replacing.

Sure, where possible.

>> do you agree to merge the size_reduction branch into master.

> Yes; sounds good.

Done.

> Yes I think that is a good plan, is there a plan of action for it?

Not yet, it's still in the "under consideration" phase.

> Although I am on holiday for two weeks from this coming weekend so will probably be off line for most of that period

Have a nice holiday :palm_tree: :relaxed:

kisvegabor, 2020-08-24:
This issue has triggered a lot of rework a lot of things (drop some widget, simplify widgets, etc.) to make LVGL smaller, even smaller than it was in v6.

The updates will be in v8 and v9. 

So I close this.

closed 2020-08-24 by kisvegabor

gmarull, 2020-08-24:
@kisvegabor do v8/v9 have an estimate release date?

kisvegabor, 2020-08-24:
We are still considering the features and ideas but I hope it will be ready end of this year.