lvgl/lvgl #1671: lv_cont_layout_grid() misjudges when put widget in new row results in space waste in right side of page

adamhan2, 2020-07-17:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

I use lv_page with pad_left and pad_inner configured and layout enabled in GRID mode. There is a margin between screen edge and boarder of lv_page. I can't make full use of the space in a page. Larger widget without enough space left in right side would result in moving to a new row. 

**To Reproduce**

I am sorry I can't submit code snippet for now. I believe it is easy to reproduce as per description. But I can figure out the route cause for this problem.  Please see in Additional Context. 

**Expected behavior**

It should allow to put widget as big as possible to make full use of the space in a container. 

**Screenshots**
[image]

**Additional context**
As shown in attached screenshot, I am expecting two widgets in a row. I have to make smaller widgets to achieve that with space waste for now. 

The Route Cause: 
  In function lv_cont_layout_grid(), the condition to check if move to a new row is as below, 
     if(act_x + inner + obj_w > w_fit) --- line 631 in lv_cont.c v7.1.0

 That is incorrect because of two reasons. 
    -- Should not mix-use the position act_x and container's width w_fit. 
         It results in space waster of pad_left of lv_page plus pad_inner of scrollable container in lv_page, plus the left margin of lv_page to screen edge. 
    -- Should not force to have a space 'inner' in right side
        It results in another pad_inner waste.

adamhan2, 2020-07-17:
I am sorry to report it in rush. I might come back to reopen it when I have more info

adamhan2, 2020-07-17:
Update root cause and possible fix solution.

embeddedt, 2020-07-17:
Can you send a pull request to the `master` branch?

kisvegabor, 2020-07-17:
Merged, thank you!

commit 75875e24 (2020-07-17): fixes #1671 lv_cont_layout_grid() fails to calculate available space in a row
  this issue results in space waste in right side of container
  the size of wasted space in each row is pad_left plus pad_inner
commit ecfef570 (2020-07-17): Merge pull request #1672 from adamhan2/fix_issue_1671

fixes #1671 lv_cont_layout_grid() fails to calculate available space in a row
closed 2020-07-17 by kisvegabor