lvgl/lvgl #1673: littlevgl7.1+rt-thread

RobotFly, 2020-07-19:
`
CC build/littlevGL7.1.0/lvgl/src/lv_draw/lv_draw_rect.o
littlevGL7.1.0/lvgl/src/lv_draw/lv_draw_rect.c: In function 'lv_draw_rect':
littlevGL7.1.0/lvgl/src/lv_draw/lv_draw_rect.c:104:1: error: insn does not satisfy its constraints:
 }
 ^
(insn 682 7070 683 33 (parallel [
            (set (reg:CC_NOOV 100 cc)
                (compare:CC_NOOV (mult:SI (reg/v:SI 6 r6 [orig:2040 size ] [2040])
                        (reg/v:SI 6 r6 [orig:2040 size ] [2040]))
                    (const_int 0 [0])))
            (set (reg/v:SI 6 r6 [orig:2040 size ] [2040])
                (mult:SI (reg/v:SI 6 r6 [orig:2040 size ] [2040])
                    (reg/v:SI 6 r6 [orig:2040 size ] [2040])))
        ]) littlevGL7.1.0/lvgl/src/lv_draw/lv_draw_rect.c:1120 38 {*mulsi3_compare0}
     (nil))
littlevGL7.1.0/lvgl/src/lv_draw/lv_draw_rect.c:104:1: internal compiler error: in extract_constrain_insn, at recog.c:2246
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
scons: *** [build/littlevGL7.1.0/lvgl/src/lv_draw/lv_draw_rect.o] Error 1

`

this error at 
`
void lv_draw_rect(const lv_area_t * coords, const lv_area_t * clip, lv_draw_rect_dsc_t * dsc)
{
    if(lv_area_get_height(coords) < 1 || lv_area_get_width(coords) < 1) return;
#if LV_USE_SHADOW
    draw_shadow(coords, clip, dsc);
#endif
`

embeddedt, 2020-07-19:
Looks like a compiler issue, not an LVGL issue.

kisvegabor, 2020-07-20:
It seems like a compiler issue to me as well. Try
1. Clean + rebuild
2. Different optimization level

RobotFly, 2020-07-20:
thanks，kisvegabor，I have solved this issue，Upgrade the version of the cross compiler gcc-arm-none-eabi-8-2018-q4-major-linux，thank you ！

closed 2020-07-20 by RobotFly

kisvegabor, 2020-07-20:
Great! :slightly_frowning_face: