lvgl/lvgl #1675: Image refresh performance

gangheivt, 2020-07-20:
In lv_img_design, when refresh the image, it will always try to redraw the background of an image. However, In most cases, Image will not have fancy effect, such as shadow, pattern, re-color etc. It just draw an image. The redraw background will almost double each image drawing time. Could we have a style to disable the background redraw?

gangheivt, 2020-07-21:
I suggest to add a style which is 32-bit integer, used as flags to aggregate several features together. This could reduce the style number as well.

kisvegabor, 2020-07-21:
I played with caching some frequently accessed style properties but it had only a very minimal effect. 

Can you measure how much time/cycles we can win if the background drawing part is commented out?

gangheivt, 2020-07-21:
@kisvegabor , I think it is the cost of cache that matters. If CPU main clock and flash access is slow, the cache cost will have more effect.  Our mainmenu has a lot of small images, that is more impacted by cache misses
[image]

kisvegabor, 2020-07-23:
> According to our hardware engineer, 1% of cache miss could almost double execution time.

That pretty interesting. Optimizing for cache hits wasn't a concern so far. Honestly, I don't know what are the principles to do so.

Can your hardware engineers give some advice in this regard?

BTW, what is the size of the I and D cache in your MCU?

gangheivt, 2020-07-25:
We use 16K I and 16K D. We found that in 48MHz MCU, even running in RAM,in 7.01 each image refresh take 200us more than 7.0. If we have multiple image, for example 20, it will consume 4ms more. This really slow down our refresh execution. I think it might worth to put some advanced feature off by default, such as shadow, patter, corner, recolor etc for image.

kisvegabor, 2020-07-27:
I see. There 3 new rectangle drawing features in v7:
- outline
- value
- pattern

We can easily make them optional with global configs. Similarly to `LV_USE_SHADOW`. background and border will be still there, but this way 4 out of 6 types can be disabled for every object type (including labels) in a clean easy to follow way.

Another (or additional) config option could be to disable all optional background drawings (label, image, arc, line meter, gauge, etc.)

What do you think?

puzrin, 2020-07-27:
If you mean "CSS outline" - it's a very rare/specific feature. Worth to drop somehow (guards/...) if it affects anything. May be, i would drop it by default, to not push wrong patterns.

`value` & `pattern` - don't understand. Where i can read about that?

puzrin, 2020-07-27:
https://docs.lvgl.io/v7/en/html/overview/style.html found here `value` & `pattern`

- `pattern` - seems need to be a part of CSS "background" property. IMO ok to disable (opionally, after move to background) if that helps.
- `value` - seems a "hand-made", not part of CSS, i'd vote for drop.

kisvegabor, 2020-07-27:
> If you mean "CSS outline" - it's a very rare/specific feature. Worth to drop somehow (guards/...) if it affects anything. May be, i would drop it by default, to not push wrong patterns.

It can be important with encoder or keypad navigation (not typical on web) to indicate the focused element.

> pattern - seems need to be a part of CSS "background" property. IMO ok to disable (opionally, after move to background) if that helps.

Yes, it's similar to background propetry in CSS but this name would be confusing here. 

>value - seems a "hand-made", not part of CSS, i'd vote for drop.

 Yes, it's handmade. Its purpose is the allow creating very light weighted text labels. Something like `::after` but can have its own font, color, alignement etc.

puzrin, 2020-07-27:
> Yes, it's similar to background propetry in CSS but this name would be confusing here.

https://developer.mozilla.org/en-US/docs/Web/CSS/background - see left column. background-image + background-position + background-repeat. Why to introduce the same functionality under different name?

> Yes, it's handmade. Its purpose is the allow creating very light weighted text labels. Something like `::after` but can have its own font, color, alignement etc.

IMO it's premature optimization, causing significant deviation from dom/css spec. Also note, nobody uses :before/:afer for texts. Those are usually applied for icons and some decor elements.

After layout rewrite this will be just a `text` node inside of `div`-like or `span`-like container.

gangheivt, 2020-07-27:
Not only image, but also other widgets got performance decrease due to new style. In our test, we used image, label, canvas and ARC. 
I would like to suggest to add a 32-bit feature mask in ext of different lv_widgets. By default, it would enable all default features. Users could choose to disable those features if necessary, or even change default features set if they feel never use those. Inside LVGL, it could minimize the style access and improve performance, try to use style get functions as less as possible. 
This implement is not compile options and might increase a little code size. However, it would let user to have flexibility to enable some specific instances of widgets to have advanced features, while still let rest(probably most) widgets to enjoy fast execution.
Another noticeable performance hit come from _lv_memcpy and _lv_memset. We used ARM library implement. It seems that ARM optimized its memcpy and memset implement. It is faster than LVGL implement. Could we make lv_memcpy and lv_memset as option?

kisvegabor, 2020-08-01:
Sorry for the late answer, I was on a holiday. 

@puzrin 
> https://developer.mozilla.org/en-US/docs/Web/CSS/background - see left column. background-image + background-position + background-repeat. Why to introduce the same functionality under different name?

I added it as a new property because I found the CSS version doesn't provide enough flexibility here. E.g. there is no `background-image-opacity` or `background-image-mix-blend-mode`.

> IMO it's premature optimization, causing significant deviation from dom/css spec. Also note, nobody uses :before/:afer for texts. Those are usually applied for icons and some decor elements.

E.g. in [lv_demo_widgets](https://lvgl.io/demos) the title of the boxes ("Basics" and "Text input") are created this way. Without it, a quite complex element structure would be required (a transparent container element with a title and a box). So it's like `::before`. So keeping at least one of them (after or before) is important in my opinion because they allow adding lightweight texts. 

However, converting them into real pseudo-elements - with their own styles - would increase their power. (It still needs to considered carefully though)

@gangheivt 
> I would like to suggest to add a 32-bit feature mask in ext of different lv_widgets. 

I feel that it'd duplicate the features. Now you can hide background/outline/pattern etc by setting their opacity to zero. Introducing another way to do this sounds messy. 
So the goal is to allow fast skipping of unset or disabled style properties. It suggests (again) the use of some kind of caching. So far I tried to cache some critical properties in `lv_style_t` but it'd be more efficient to cache in `lv_style_list_t` (the collection of styles of a part of an object). Basically style properties can change only when a new style is added, removed, one of its properties has changed or the parent has changed. So we can cache the final presence of specific, critical properties (after considering all styles, inheritance, etc).  
What do you think?

> Another noticeable performance hit come from _lv_memcpy and _lv_memset. We used ARM library implement. It seems that ARM optimized its memcpy and memset implement. It is faster than LVGL implement. Could we make lv_memcpy and lv_memset as option?

Sure, there is no problem with that.

puzrin, 2020-08-02:
> I added it as a new property because I found the CSS version doesn't provide enough flexibility here. E.g. there is no `background-image-opacity` or `background-image-mix-blend-mode`.

- image can have alpha channel
- `div` can have opacity

IMO that's enough.

`background-image-mix-blend-mode` - not sure this required everywhere. Why CSS works fine without it?

> E.g. in [lv_demo_widgets](https://lvgl.io/demos) the title of the boxes ("Basics" and "Text input") are created this way. Without it, a quite complex element structure would be required (a [...]

IMO you reverse result and reason. If you can use :pseudo elements, that does not means it's a right way of use. I'd recommend to avoid that. Current architecture miss `text` node (most close is `label`). Solve explicit problem first.

> Without it, a quite complex element structure would be required (a transparent container element with a title and a box)

Year... "why those idiots invented dom and css, when everyone can imagine much better?" :)

embeddedt, 2020-08-02:
> Current architecture miss text node (most close is label). Solve explicit problem first.

I suspect there was a deliberate reason why that was avoided (probably performance or code size).

puzrin, 2020-08-02:
> I suspect there was a deliberate reason why that was avoided (probably performance or code size).

May be. But since architecture changed, it worth to clarify this point too, instead of diverge from CSS.

I suggest to stay with simplest case - list of chilren with single text node per container (without `aaa <b>bbb</b> ccc`). That will give simple implementaion, and following CSS spec without problems.

Also, at first glance, this should allow extend render without breaking upper abstractions of data.

puzrin, 2020-08-02:
> So the goal is to allow fast skipping of unset or disabled style properties. It suggests (again) the use of some kind of caching. So far I tried to cache some critical properties in `lv_style_t` but it'd be more efficient to cache in `lv_style_list_t` (the collection of styles of a part of an object). Basically style properties can change only when a new style is added, removed, one of its properties has changed or the parent has changed. So we can cache the final presence (considering all styles, inheritance, etc) of specific, critical properties.
> What do you think?

Usually, the most difficult part of cache is invalidation, when exists. So, in context of tree traverse, i'd suggest to cache state [styles] of parents. That will reduce scan iterations significantly, but without need of invalidaion.

gangheivt, 2020-08-02:
@kisvegabor , the following style access list is a typical image refresh, I turn off many advanced features, AND set many opa to transparent:
lv_obj_get_style_radius
lv_obj_get_style_bg_opa
lv_obj_get_style_border_width
lv_obj_get_style_outline_width
lv_obj_get_style_pattern_image
lv_obj_get_style_shadow_width
lv_obj_get_style_value_str
lv_obj_get_style_border_post
lv_obj_get_style_transform_zoom
lv_obj_get_style_transform_angle
lv_obj_get_style_pad_left
lv_obj_get_style_pad_right
lv_obj_get_style_pad_top
lv_obj_get_style_pad_bottom
lv_obj_get_style_clip_corner
lv_obj_get_style_image_opa
lv_obj_get_style_image_recolor_opa
lv_obj_get_style_image_recolor

This is ONLY one Image I tried to refresh. If there are many images on the screen, the style access will slow down the refresh rate significantly. Even I set OPA to transparent, still it need to check the OPA value using style access. 

To refresh this image, I only need to access angle and zoom value in image ext.  If one access check might need, say 250 assembly instructions (This depends on how many parents/grand parents of the objects), it will require 4500 instructions, and in 48MHz MCU, will require 93.75us.  If 20 images are on screen, it will consume 1.875ms.  The original refresh rate of 30 frame/sec will drop to 28.4 frame/sec.  In our real test, each image refresh will add 200us more using style access in 7.01. 

There are also other style access in our application running. Without optimization, our typical refresh rate drop is between 10-20% in 7.01 even all images are running in ROM/RAM. If we added flash cache miss, the impact will be even more, and usually is between 20-30%.

kisvegabor, 2020-08-03:
@puzrin 
Regarding the "text node" thing. So far the similarity with HTML/CSS wasn't a goal an I considered the rectangles as basic blocks and not the texts. However, I see that by merging label (and even image in some way?) features into the basic object can highly reduce memory usage because this way the current "button+label" cases are simplified to "button" (half memory).

Another example: now lists are created from "button+label+icon" but they could be "button with text + icon with 'after' ". It'd reduce the memory used by 66%. 

So I see a great value in introducing a text property on the base object but I suggest using a dedicated issue for these things to keep the discussion clear. Could you open an issue for it?

@gangheivt 
I implemented a new way of style caching in `feat/style-cache` branch. Key points:
- no extra memory is used because the cache fits into 3 bytes in [lv_style_list_t](https://github.com/lvgl/lvgl/blob/feat/style-cache/src/lv_core/lv_style.h#L232) that were padded so far
- the idea is: a lot of properties have default value but they still need to be read frequently. So the cache stores if a property has non-default value. 
- some critical properties (almost all you listed) are cached 
- with `lv_demo_widgets` I measured 40% cache hit ratio (i.e. 40% fewer style queries). On a simple image it should be almost 100%.  

Could you measure the effect of this new feature?

gangheivt, 2020-08-04:
@kisvegabor , We are working on it, it might take some time, as we are still using 7.01 with our own optimization.

kisvegabor, 2020-08-05:
No problem, look forward to hearing your feedback.

Meanwhile, I added [LV_MEMCPY_MEMSET_STD](https://github.com/lvgl/lvgl/blob/1f0a4918f21e8ae44f6825ed26eb4a4c8c84a6f7/lv_conf_template.h#L102) config option to use standard memcpy and memset.

gangheivt, 2020-08-12:
@kisvegabor , we have tried style cache, it still about 8% slower than our current implement. We put some features flags(such as round corner, background OPA etc) in image ext, and disable unused. 
Since we also implemented GPU Async, this delay is compensated by GPU enhancement.  This delay is not in the critical path. 
In general, we could improve our refresh performance with style cache you have implemented. I suggest you to merge the implement, and close this issue.

kisvegabor, 2020-08-12:
Thanks for the feedback!

Do you know the overall speedup as well?

gangheivt, 2020-08-13:
In our system, In  the original 7.01, each image refresh took about 800us in style access, Our own resolution reduced to 350us. With your implement, it is around 380us.  Please note that there are some overhead for performance data collection.

kisvegabor, 2020-08-13:
Wow, not bad. ^^ 

I merged it into the `dev` branch. Thank you for testing!

closed 2020-08-13 by kisvegabor