lvgl/lvgl #1676: abnormal tabview on STM32F429

zhwxChos, 2020-07-21:
Hi,
1、platform: STM32F429
2、liitleVGL  7.2.0
3、Everything works fine on the simulator.
4、when clicking the tabview tab, sometimes the tab backgroud color and the font clolor will be changed,not recover. 
And long press,the tabview page background color will be changed too , beacuse it's status is act?
This looks a bit abnormal!
  
[image]

kisvegabor, 2020-07-21:
Please send a minimal code snippet to reproduce.

Do you see the issue with fewer tabs as well?

zhwxChos, 2020-07-24:
@kisvegabor in  lv_tabview_add_tab , code(line no :233-242)  is repeated with the following code(line no:251-282)。I think code (line no:233-242) is waste.
[image]

zhwxChos, 2020-07-24:
> Please send a minimal code snippet to reproduce.
> 
> Do you see the issue with fewer tabs as well?

fewer tabs aslo has this issue.  It looks like the background color after getting the focus,and will not disappear.

kisvegabor, 2020-07-24:
I've removed the duplicated lines, thanks.

But a code snippet to reproduce the issue is still required to see what's going on.

pete-pjb, 2020-07-24:
Hi @zhwxChos,

Just a note I had issues of this nature with the STM32429I-EVAL Board from ST using their STemWin.  The problem turned out to be the ST touch screen driver not generating touch release events intermittently and the touch control chip locking up intermittently for no apparent reason.  In the end I re-wrote the driver and implemented interrupts.  I don't remember all the details now it was a few years ago.  If you are using ST's drivers and the STMPE1600 touch control chip it may be worth a look there.

Hope that helps.

Kind Regards,

Pete

zhwxChos, 2020-07-25:
Hi，@pete-pjb
Thank you for sharing this , I will try it .

zhwxChos, 2020-08-11:
Hi @kisvegabor,
I can't solve this problem。。。

Why does the background and the cursor flash together？

I converted the video to gif format, About 54M,
Url : https://pan.baidu.com/s/1ffzRD1bkU93RAnC0S3kDOQ    Code: hjnu

And code is simple.
`lv_obj_t* par = lv_scr_act();
m_contentPage = lv_page_create(par, NULL);
	lv_obj_set_size(m_contentPage, lv_obj_get_width_fit(par) - 5, 220);
	lv_obj_set_pos(m_contentPage, 3, 20);
	lv_obj_set_click(m_contentPage, false);
	
	lv_obj_t* text = addLabel(m_contentPage, NULL, "A");
	lv_label_set_align(text, LV_LABEL_ALIGN_LEFT);
	lv_obj_set_pos(text, 5,5);

	lv_obj_t* ta = addEdit(m_contentPage, NULL, "B");
	lv_obj_set_width(ta, 70);
	lv_obj_align(ta, text, LV_ALIGN_OUT_BOTTOM_LEFT, 2, 2);
	lv_obj_set_event_cb(ta, onFocusedTa);

	text = addLabel(m_contentPage, NULL, "C");
	lv_obj_align(text, ta, LV_ALIGN_OUT_BOTTOM_LEFT, -2, 2);

	ta = addEdit(m_contentPage, ta,"");
	lv_obj_align(ta, text, LV_ALIGN_OUT_BOTTOM_LEFT, 2, 2);

	text = addLabel(m_contentPage, NULL, "D");
	lv_obj_align(text, ta, LV_ALIGN_OUT_BOTTOM_LEFT, -2, 2);

	ta = addEdit(m_contentPage, ta, "");
	lv_obj_align(ta, text, LV_ALIGN_OUT_BOTTOM_LEFT, 2, 2);`

kisvegabor, 2020-08-12:
Quick question: do you use the latest version of LVGL from the master branch?

zhwxChos, 2020-08-12:
> Quick question: do you use the latest version of LVGL from the master branch?
V7.3.0

kisvegabor, 2020-08-13:
Very strange. Please enable all `LV_USE_ASSERT_...` options in `lv_conf.h`. Hopefully they catch something.

zhwxChos, 2020-08-14:
@kisvegabor 
My display interface is LTDC , and SDRAM is used.
I used **Two screen-sized buffers**  ,display blinks.     **One buffer** ,display normal.

buf_1,buf_2 are in SDRAM. 

[image]

kisvegabor, 2020-08-14:
How does your `disp_flsuh` function look like and where do you call `lv_disp_flush_ready`?

zhwxChos, 2020-08-14:
@kisvegabor 
`static void disp_flush(lv_disp_drv_t * disp_drv, const lv_area_t * area, lv_color_t * color_p)
{
	LTDC_Color_Fill(area->x1,area->y1,area->x2,area->y2,(uint16_t*)color_p);
	lv_disp_flush_ready(disp_drv);
}
void lv_port_disp_init(void)
{
	static lv_disp_buf_t disp_buf;

	lv_disp_buf_init(&disp_buf, color_buf, NULL, COLOR_BUF_SIZE); 

	lv_disp_drv_t disp_drv;                         
	lv_disp_drv_init(&disp_drv);                 

	disp_drv.hor_res = lcdltdc.width;
	disp_drv.ver_res = lcdltdc.height;
	disp_drv.flush_cb = disp_flush;

	disp_drv.buffer = &disp_buf;

#if LV_USE_GPU  //0
	disp_drv.gpu_blend_cb = gpu_blend;

	disp_drv.gpu_fill_cb = gpu_fill;
#endif
	lv_disp_drv_register(&disp_drv);
}

`

kisvegabor, 2020-08-17:
Ah, I see, the problem.  With true double buffering you need to change the frame buffer address to `color_p` in `disp_flush`.

LVGL swaps buffers automatically and always give you the next buffers.

zhwxChos, 2020-08-17:
@kisvegabor Thank you.
Can you give me an example ?

embeddedt, 2020-08-17:
@zhwxChos You just need to find where in your display initialization code the framebuffer address is set, and re-use that code in `disp_flush`, setting the address to `color_p` instead of whatever it was before.

zhwxChos, 2020-08-17:
@embeddedt Not very clear, show your code,^_^

In the function LTDC_Color_Fill, calc the address of color_p ,and assign it to DMA2D->OMAR,then transmit.

framebuffer address is 0XC0000000 in SDRAM, disp_buf->buf1 address is (0XC0000000 + screen size * color depth) , 
Can the address of disp_buf->buf1 be directly set to the address of framebuffer ? Thereby reducing memory copy

embeddedt, 2020-08-17:
> framebuffer address is 0XC0000000 in SDRAM

That's the part you need to change in `disp_flush`.

closed 2020-08-28 by zhwxChaos