lvgl/lvgl #1678: How to cancel the widget operation of the background layer

RanBiron, 2020-07-21:
When I create a message box, the message box will be in the foreground layer. What I want is that all the widgets in the background layer cannot be clicked or operated. How do I do it?

pete-pjb, 2020-07-21:
Hi @RanBiron,

We normally use the [forum](https://forum.lvgl.io) for questions, whilst we use Github for development discussions etc.  Please can you post in the forum next time please. 😄 Much appreciated thanks.

In answer to your question please take a look at the docs [here](https://docs.lvgl.io/v7/en/html/widgets/msgbox.html#modal) there is a code sample to get you started.  If you need any further assistance don't hesitate to ask.

Kind Regards,

Pete

kisvegabor, 2020-07-21:
I closed the issue. Please ask this question in the Forum.

closed 2020-07-21 by kisvegabor