lvgl/lvgl #1681: Garbage can be seen at the edge of switch

lgl88911, 2020-07-22:
When the switch is switched, garbage can be seen at the edge  during the animation。Attached video display phenomenon。
In addition to switch, other widgets can also see this garbage when performing animation effects, such as dorp list and button。

I am not sure if this is the lvgl problem or the LCD screen problem. Does anyone have a similar problem? If you can get your guidance, I would be very grateful。

I using 16bit RGB interface LCD Hardware refresh rate is 60fps, The resolution is 800*480.
lvgl is configured as double buffer, refresh time LV_DISP_DEF_REFR_PERIOD = 25. Execute handle task every 10ms.
```
	while(1){
		s64_t start = k_uptime_get();
		lv_task_handler();
		start = k_uptime_get() - start;
		if(start > 0)
			printk("flush %lld\n", start);
		lv_tick_inc(10);
		k_sleep(K_MSEC(10));
	}
```
[VID_20200723_001040.zip](https://github.com/lvgl/lvgl/files/4961156/VID_20200723_001040.zip)

pete-pjb, 2020-07-23:
Hi @lgl88911,

Please can you use the [forum](https://forum.lvgl.io) for such questions in future, you can log in to the forum with your Github account, we use Github issues to discuss development related material. Thank you. 😃 

In my experience this type of issue is often caused by a cache related issue.  So I would check the cache configuration for the video buffer first.  What hardware platform are you using?  Please reply in the [forum](https://forum.lvgl.io) to continue the discussion if you require further assistance and I will close this issue here.

Kind Regards,

Pete

closed 2020-07-23 by pete-pjb

lgl88911, 2020-07-23:
@pete-pjb Thanks for your reply .
 I create a topic https://forum.lvgl.io/t/garbage-appears-in-the-horizontal-direction-of-the-widgets/2813 to disscus the issue.
By the way, there is flush cache before each render, maybe other reasons cause this problem