lvgl/lvgl #1683: How to draw an initially unknown number of lines?

denruss, 2020-07-23:
For example a simple function that draws a simple line

`void drawLine(int16_t x0, int16_t y0, int16_t x1, int16_t y1)
{
	static lv_point_t line_points[2] ;

	line_points[0].x = x0;
	line_points[0].y = y0;
	line_points[1].x = x1;
	line_points[1].y = y1;

    /*Create style*/
    static lv_style_t style_line;
    lv_style_init(&style_line);
    lv_style_set_line_width(&style_line, LV_STATE_DEFAULT, 8);
    lv_style_set_line_color(&style_line, LV_STATE_DEFAULT, LV_COLOR_BLUE);
    lv_style_set_line_rounded(&style_line, LV_STATE_DEFAULT, true);

    /*Create a line and apply the new style*/
    lv_obj_t * line1;
    line1 = lv_line_create(lv_scr_act(), NULL);
    lv_line_set_points(line1, line_points, 2);     /*Set the points*/
    lv_obj_add_style(line1, LV_LINE_PART_MAIN, &style_line);     /*Set the points*/
    //lv_obj_align(line1, NULL, LV_ALIGN_CENTER, 0, 0);}
}`

but if you call drawLine() several times, then only the last line is drawn.

How can I draw in advance an unknown number of lines?

embeddedt, 2020-07-23:
Allocate `line_points` instead of using a static array:

```c
lv_point_t * line_points = lv_mem_alloc(sizeof(lv_point_t) * 2);
```

You'd of course need extra logic if you wanted to be able to remove the lines afterwards.

excitedbox, 2020-08-10:
This is coming from a php background so it may not apply to C where I am just a noob. Could you change all references to line1 to a variable and change it's value to line1, line2, line3 etc on each call? That way you are creating a style for each line instead and don't need additional logic for removing each line because you can use the same variable value to remove them. Is that correct or am I way off the mark?

essentially a function that is draw_line(linename){ your code here  } in a for loop you could then use line+your counter value as the line name to draw a bunch of lines.

kisvegabor, 2020-08-11:
@excitedbox 
It's better to store all points in an array then creating line object for every line segment because an object needs about 100 bytes. For 20 segments it's 20 kB. 

Let me know if I misunderstood your idea.

excitedbox, 2020-08-11:
I think I misunderstood what he was trying to do. I thought he was drawing multiple lines and not line segments of the same line. 
You are right an array is more efficient than having multiple variables. I just meant replacing hard coded line "names" with dynamic names. So line1 would be *line[ ] to store a name for each line in a pointer array. But don't array lengths need to be predefined so not knowing the number of lines presents a problem?

kisvegabor, 2020-08-12:
> But don't array lengths need to be predefined so not knowing the number of lines presents a problem?

If you need 10 points yout can dynamically allocate an array for 10 points and can apply them like this:
```c
lv_line_set_points(line1, line_points, 10);  
```

So you can pass the current length of the array.

excitedbox, 2020-08-12:
Ok, I think I get it now thanks.

I was just going through the examples to copy some code and realized it is kind of hard to tell where one widget starts and the next one ends. Would you like me to add some comments to make it easier to see what belongs together?

kisvegabor, 2020-08-13:
> Would you like me to add some comments to make it easier to see what belongs together?

Sure, it'd be awesome!

Can you send a PR  with only one file to see how your updates look?

kisvegabor, 2020-09-01:
I lack of feedback from the OP I close this issue.

closed 2020-09-01 by kisvegabor