lvgl/lvgl #1685: Should usleep() in example code be replaced by nanosleep?

detly, 2020-07-24:
According to the [`usleep(3)` manual page](https://man7.org/linux/man-pages/man3/usleep.3.html):

> POSIX.1-2001 declares this function obsolete; use nanosleep(2) instead.  POSIX.1-2008 removes the specification of usleep().

You currently have to jump through [a few `#define` hoops](https://stackoverflow.com/a/55860859/188535) to use `usleep()` too. It might be worthwhile to use [`nanosleep()`](https://www.gnu.org/software/libc/manual/html_node/Sleeping.html) instead:

```c
#include <time.h>

static void * tick_thread(void * arg)
{
    (void) arg;

    const struct timespec period = {
        .tv_sec = 0,
        .tv_nsec = 5000000UL,
    };

    struct timespec remaining;

    while(1) {
        nanosleep(&period, &remaining);
        lv_tick_inc(5);
    }

    return NULL;
}
```

There's a note in [the Glibc docs](https://www.gnu.org/software/libc/manual/html_node/Sleeping.html) about how to use the remaining time parameter:

> Resist the temptation to implement a sleep for a fixed amount of time by using the return value of sleep, when nonzero, to call sleep again. This will work with a certain amount of accuracy as long as signals arrive infrequently. But each signal can cause the eventual wakeup time to be off by an additional second or so. Suppose a few signals happen to arrive in rapid succession by bad luck—there is no limit on how much this could shorten or lengthen the wait.

> Instead, compute the calendar time at which the program should stop waiting, and keep trying to wait until that calendar time. This won’t be off by more than a second. With just a little more work, you can use select and make the waiting period quite accurate. (Of course, heavy system load can cause additional unavoidable delays—unless the machine is dedicated to one application, there is no way you can avoid this.) 

But for the sake of demo code, I think this would overcomplicate things for not much gain. Plus (a) this was a problem in existing demo code anyway, and (b) is anyone running LVGL in a program with high use of signals?

embeddedt, 2020-07-24:
Personally, I'd still prefer the simplicity of `usleep`, especially since this is meant to be an example.

While it's deprecated, I don't expect it to be removed from any libc implementation any time soon, since there are a lot of legacy applications out there that probably make use of it.

@kisvegabor @pete-pjb What do you think?

pete-pjb, 2020-07-24:
Yes I agree with you @embeddedt, its just an example and not a real system and would bring unnecessary confusion to some I am sure.

detly, 2020-07-24:
I'm happy to close this then. I don't have strong feelings about it, it was just something I noticed.

closed 2020-07-24 by detly

kisvegabor, 2020-07-27:
Sorry, for the late replay.

I also agree to keep it for a while.