lvgl/lvgl #1686: LV_STATE_FOCUSED | LV_STATE_PRESSED doesn't work

shadowninja69, 2020-07-24:
**Describe the bug**

Setting style with both LV_STATE_FOCUSED | LV_STATE_PRESSED states doesn't work. It only applies to LV_STATE_PRESSED.

**To Reproduce**

You can reproduce this in lv_demo_keypad_encoder with adding the following code

```
    static style = {0};
    lv_style_init(&style);
    lv_style_set_border_color(&style, LV_STATE_FOCUSED | LV_STATE_PRESSED, LV_COLOR_RED);

    selector_objs.btn = lv_btn_create(parent, NULL);
    lv_obj_t* label = lv_label_create(selector_objs.btn, NULL);
    lv_label_set_text(label, "Button 1");
    lv_obj_add_style(selector_objs.btn, LV_BTN_PART_MAIN, &style);

    selector_objs.btn2 = lv_btn_create(parent, NULL);
    lv_obj_t* label2 = lv_label_create(selector_objs.btn2, NULL);
    lv_label_set_text(label2, "Button 2");
    lv_obj_add_style(selector_objs.btn2, LV_BTN_PART_MAIN, &style);
```

You don't need the second button, I just added it so that switching the focus between the 2 buttons makes it more obvious.

When only using LV_STATE_FOCUSED or LV_STATE_PRESSED exclusively, it works. But when using both, it only gets applied to LV_STATE_PRESSED.

**Expected behavior**

I would expect the style to be applied to both focused and pressed state, but it only got applied to pressed state.

embeddedt, 2020-07-24:
The style will be applied when the button is both pressed and focused. That's what happens when you combine states.

If you want it to be applied independently to both, you should call the function twice, with a different state argument each time.

shadowninja69, 2020-07-24:
Oh that's what it means. thx for explaining.

closed 2020-07-24 by shadowninja69