lvgl/lvgl #1687: btnmatrix buttons don't respond to LV_STATE_PRESSED style

shadowninja69, 2020-07-24:
**Describe the bug**

Tried to set a LV_STATE_PRESSED style to button matrix buttons, nothing happened. Tried LV_BTN_STATE_PRESSED, didn't work either.

**To Reproduce**

Here's a simple style and a btnmatrix copied from the doc that reproduces this in `lv_demo_keypad_encoder`

```
    static style = {0};
    lv_style_init(&style);
    lv_style_set_border_color(&style, LV_BTN_STATE_PRESSED, LV_COLOR_RED);

    // example copied from lv btnmatrix docs
    selector_objs.btnm1 = lv_btnmatrix_create(parent, NULL);
    lv_btnmatrix_set_map(selector_objs.btnm1, btnm_map);
    lv_btnmatrix_set_btn_width(selector_objs.btnm1, 10, 2); /*Make "Action1" twice as wide as "Action2"*/
    lv_btnmatrix_set_btn_ctrl(selector_objs.btnm1, 10, LV_BTNMATRIX_CTRL_CHECKABLE);
    lv_btnmatrix_set_btn_ctrl(selector_objs.btnm1, 11, LV_BTNMATRIX_CTRL_CHECK_STATE);
    lv_obj_align(selector_objs.btnm1, NULL, LV_ALIGN_CENTER, 0, 0);
    lv_obj_add_style(selector_objs.btnm1, LV_BTNMATRIX_PART_BTN, &style);
```

This was what confused me about #1686

kisvegabor, 2020-07-27:
In `lv_style_set_border_color(&style, LV_BTN_STATE_PRESSED, LV_COLOR_RED);` you should use `LV_STATE_PRESSED`.

shadowninja69, 2020-07-27:
> In `lv_style_set_border_color(&style, LV_BTN_STATE_PRESSED, LV_COLOR_RED);` you should use `LV_STATE_PRESSED`.

Like I mentioned, neither worked. But it seemed the issues was I had to use `LV_STATE_FOCUSED | LV_STATE_PRESSED` instead. Which brings the question, do I have to manually set styles for all combined states?

shadowninja69, 2020-07-27:
Reopening and re-titled the issue, as `LV_STATE_FOCUSED | LV_STATE_CHECKED` and `LV_STATE_FOCUSED | LV_STATE_DISABLED` combined states still doesn't seem to work

kisvegabor, 2020-07-27:
> Like I mentioned, neither worked. 

Oh, sorry, my bad. 

I've fixed it in 06d7a6d0.

shadowninja69, 2020-07-27:
Thanks! It worked!
The other issue I had was `LV_STATE_CHECKED` doesn't seem to allow setting a border. Would this also be fixable?

kisvegabor, 2020-08-01:
In the default material theme the border width of checked state is zero. So e.g. `lv_obj_set_style_local_border_width(btnm1, LV_BTNMATRIX_PART_BTN, LV_STATE_CHECKED, 3); ` makes the border appear. 
`

shadowninja69, 2020-08-01:
ah so that's why. thanks a lot!

closed 2020-08-01 by shadowninja69