lvgl/lvgl #1689: Compile failed

weixiongmei, 2020-07-27:
lvgl/src/lv_themes/lv_theme_material.c:214:56: error: bitwise comparison always evaluates to false [-Werror=tautological-compare]
     if((theme.flags | LV_THEME_MATERIAL_FLAG_NO_FOCUS) == 0)lv_style_set_border_color(&styles->bg, LV_STATE_FOCUSED, theme.color_primary);

Pull revision: 834b498a

kisvegabor, 2020-07-27:
It's strange. I can1t see why is it evaluated as false. `flags` is `uint32_t` and `LV_THEME_MATERIAL_FLAG_NO_FOCUS` is `0x20`.

embeddedt, 2020-07-27:
Looks like it's an enum member: https://github.com/lvgl/lvgl/blob/ed10c1da672d2865ec37fb6229f9bc646086d983/src/lv_themes/lv_theme_material.h#L27

Is a C compiler allowed to reorder enums for optimization purposes? It might be worth confirming at runtime that the value really is `0x20`.

pete-pjb, 2020-07-27:
Hi @kisvegabor ,

If I am understanding your intentions correctly might it be better to have:

```c
if(!(theme.flags &  LV_THEME_MATERIAL_FLAG_NO_FOCUS))lv_style_set_border_color(&styles->bg, LV_STATE_FOCUSED, theme.color_primary);
```
as opposed to:
```c
if((theme.flags | LV_THEME_MATERIAL_FLAG_NO_FOCUS) == 0)lv_style_set_border_color(&styles->bg, LV_STATE_FOCUSED, theme.color_primary);
```

The original bit-wise operation ```theme.flags | LV_THEME_MATERIAL_FLAG_NO_FOCUS``` will always yield an answer of true so I agree with the compiler on this one.

Hope that makes sense.

Kind Regards,

Pete

embeddedt, 2020-07-27:
Ah, that makes sense. I should have looked at the operator more closely.

kisvegabor, 2020-07-27:
Ah, I also overlooked it. :smile: Thanks, @pete-pjb

commit 2172ec93 (2020-07-27): material theme minor fix

Fixes #1689
closed 2020-07-27 by kisvegabor