lvgl/lvgl #1690: lv_style_get_bg_color does not exist but is used in documentation

thedjnK, 2020-07-27:
**Describe the bug**

lv_style_get_bg_color function does not exist but is referred to in documentation

**To Reproduce**

According to https://docs.lvgl.io/v7/en/html/overview/style.html, this is valid code:

    res = lv_style_get_bg_color(&style1, LV_STATE_PRESSED, &color);

**Expected behavior**

Function to exist/application to compile

**Actual behavior**

    error: 'lv_style_get_bg_color' was not declared in this scope

**Additional context**

Have grep'd through the source and got 0 results for "lv_style_get_bg_color"

nicholaskunes, 2020-07-27:
```lv_obj_get_style_bg_color``` does exist.

kisvegabor, 2020-08-02:
Hi, 

Thanks for pointing this out.

I've updated the docs with the correct function names.

closed 2020-08-03 by kisvegabor