lvgl/lvgl #1696: LV_DPI parameter won't change anything

mahyar1284, 2020-07-31:
I'm simulating a 7inch 1024x600 (i.e. the dpi is about 170) lcd on visual studio using lvgl simulator tamplete but whn I change dpi to neather 10 nor 300 nothing changes. I want to scale up everything on my lcd.
[image]
this is my designed user interface but changing the LV_DPI won't change the size of texts and...
`	
// made in iran parameter
lv_obj_t* mii_txt = lv_label_create(lv_scr_act(), NULL);
lv_label_set_text(mii_txt, "Designed and Manufactured in I.R.Iran: ");
lv_obj_align(mii_txt, temp_txt, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 10);
`
for example this is a part of my code. and I have changed LV_DPI in LV_Conf file to many different numbers. but nothing has changed

pete-pjb, 2020-07-31:
Hi @mahyar1284 ,

We normally use the [LVGL forum](https://forum.lvgl.io) for LVGL questions, please can you post there in future, Github issues are reserved for development discussions. You can login to the [forum](https://forum.lvgl.io) with your Github account. Thank you. :smiley:

As far as I am aware the LV_DPI parameter does not have anything to do with the the screens 'Dots per Inch'  I believe it is an internal value that alters padding sizes around various objects when they are drawn in LVGL.

I don't think there is an easy way to simulate the DPI characteristics of a screen because the screen you are using for the simulation kind of dictates what you are going to see.  I would expect if you set the simulators horizontal and vertical resolution in lv_conf.h
```c
/* Maximal horizontal and vertical resolution to support by the library.*/ 
#define LV_HOR_RES_MAX          (1024)
#define LV_VER_RES_MAX          (768)
```
 to match your intended display you should get a pretty reasonable representation of what you are going to see on your display.

I hope that helps.

I will close this issue here for now, if you want to continue the discussion please post in the forum.

Kind Regards,

Pete

closed 2020-07-31 by pete-pjb

embeddedt, 2020-07-31:
> As far as I am aware the LV_DPI parameter does not have anything to do with the the screens 'Dots per Inch' I believe it is an internal value that alters padding sizes around various objects when they are drawn in LVGL.

That's correct. It will alter, to some extent, the sizes of objects in the demos (provided that they calculate their size by multiplying a constant times `LV_DPI`) but it doesn't automatically scale fonts. To do that, you should choose a larger font size in `lv_conf.h`.