lvgl/lvgl #1698: lv_obj_align_origo functions: "origo" has no meaning

upbeat27, 2020-07-31:
In the release notes for 7.2.0 I noticed the addition of some functions, ```lv_obj_align_origo``` and related.
It looks like this is intending to be related to "origin" but I have never seen it abbreviated as "origo".
I would suggest for the sake of readability and comprehension, that these function names be changed (sooner rather than later!)

kisvegabor, 2020-08-03:
Hi,

They can be used to align the center point of the object. Can you suggest a good name (short, still descriptive)?

upbeat27, 2020-08-03:
'''lv_obj_align_origin'''
I get what the functions do, it is the "origo", or more specifically the last "o" there that makes the naming weird.

kisvegabor, 2020-08-04:
Oh god... In Hungarian we the center of a coordinate system "origo". I thought its international... :sweat_smile: 
I wonder how it wasn't mentioned so far.

Anyway, even "origin" is misleading as the origin of the coordinate system is in the top left corner of the object. So it's better to look for a totally new name. What about `lv_obj_align_mid`?

upbeat27, 2020-08-04:
Well that explains it 🙃
You're right about origin too. I thought that was where origo was coming from...
"Mid" is shorter than "center", which is probably the other option.

kisvegabor, 2020-08-04:
"center" might be confusing because this work is used in `LV_ALIGN_CENTER` too. 

I'll rename them to mid soon.

upbeat27, 2020-08-04:
OK.
While you're at it... I noticed a similar issue: ```lv_chart_clear_serie```. Should be ```lv_chart_clear_series```.

closed 2020-08-04 by upbeat27

kisvegabor, 2020-08-04:
I fixed both here: bd4db19a

Thanks!