lvgl/lvgl #1699: [lv_font_conv] Using a large xxx.ttf file causes a memory exception error

SmartAnda, 2020-08-01:
**Using the lv_font_conv, I found an exception error(about the memory request exception), which occurs when the xxx.ttf file is large(e.g. 16MB)**

--------------------------------------------------------------------------------------------------------------------------------
collect_font_data.js ==> collect_font_data
fonts_freetype[source_path] = ft_render.fontface_create(source_bin, args.size);
index.js ==> fontface_create
  let error;
  let face = {
    ptr: 0,
    font: m._malloc(source.length) ==> My test info: source.length = 13457164, an exception is raised at this location.
  };
--------------------------------------------------------------------------------------------------------------------------------

[image]
[image]

Looking forward to your reply. Thank you in advance. 

@kisvegabor @puzrin

puzrin, 2020-08-01:
Please provide ALL details how to reproduce, step-by-step.

SmartAnda, 2020-08-01:
Thank you very much for your quick reply.

The reproduce method:

node lv_font_conv.js --font C:\SourceHanSansK-Regular.ttf --range 0x20-0x7F --size 24 --format lvgl --bpp 4 lv_font_test.c

[lv_font_ttf_test.zip](https://github.com/lvgl/lvgl/files/5010295/lv_font_ttf_test.zip)

@puzrin

puzrin, 2020-08-01:
@SmartAnda  Tried as you said. Can not reproduce in Ubuntu 18.04LTS (both with node 12 & 14). Please, do this things and tell me result:

- What node/win version do you use? (`node -v`)
- If node is not latest - upgrade to last v12 or v14.
- Try web convertor in Chrome/Firefox (with different webassenbly engine)

SmartAnda, 2020-08-01:
@puzrin I just tested this on my another computer and it was fine (the node.exe version is v12.18.2, as the image shows)
[image]

I will confirm again on next monday.

Best Regards.

SmartAnda, 2020-08-03:
Hey, @puzrin 
I have solved this problem now.

I just updated the lv_font_conv from github repo, and I found that the ft_render file has been updated.
The important point => [ freetype: allow wasm memory growth, #29 ]

Now everything is good.

Best Regards.

closed 2020-08-03 by SmartAnda