lvgl/lvgl #1701: text node (or text property) in/with base objects

puzrin, 2020-08-03:
With upcoming changes ("moving into CSS direction") in seems obvious that "text containers" are missed (labels are too specific).

- CSS is applied to DOM
- DOM is expected to have some basic functionality
- "layout rewrite" cares about `display:block` things, but no cares about `display:inline` things.
- `text` nodes are missed. `labels` are specific "block" things (due lack of `div` & `text`).

It would be more correct to fix data layer, adding `text` node. For very simple case this can be `.text` property of `lv_obj`, but it can cause too much breaking changes in future. Partial implementation of web's text node should have the same cost without side effects.

**What is suggested:**

- implement "text" object - simple inline-kind container with only one property - text string.
- in layout, implement only simpliest case for `<div>` text - single text child (without `aaa <b>bbb</b> ccc`).

kisvegabor, 2020-08-04:
One initial question...

Let's say I have this HTML snippet:
```html
<div>
First text node
<button>button</button>
Second text node
</div>
```

In LVGL we could do something like this:
```c
lv_obj_t * div = lv_obj_create(lv_scr_act(), NULL);
lv_obj_add_text(div, "First text node");
lv_obj_t * btn = lv_obj_create(btn, NULL);
lv_obj_add_text(btn, "Button");
lv_obj_add_text(div, "Second text node"); 
```

So to follow HTML's working mechanism we need to be able to have multiple text nodes and register their order relative to the "normal" children. If so it looks like a big additional complexity for a not-sure-we-need use case. 

I don't want to do the same rounds again - "what if change differ only a little from HTML/CSS" - because I understand the value of using something proven. But let's validate if it's a common or good practice in HTML to do things as I've shown in the snippet.

puzrin, 2020-08-04:
Answer is simple: do subset, not full copy. In this concrete case - allow multiple children, but render only single text node if exists. Later this can be extended or left as is - but you will have guarantee of correctness.

As far as i understand, in v8 you will bring equivalent of `<div>` with allowed children for inner content. All you need is support case with single text node as child, ~ as you do in label.

puzrin, 2020-08-04:
Side note. You CAN emulate single text node via DIV's `.text` property. But, to minimize deviations from DOM/HTML/CSS, need to keep in mind:

- `.text` can NOT be part of background. It's the content, not decoration (content may define size [height] of DIV).
- If text property exists, children CAN NOT be added (because semantics become undefined).
- Need to remember, `.text` property MUST behave like single child text node (this is too fragile, attractive for "home-made" divergences).
- That will have to be dropped, if anyone decide to extend inline content (add bold, italic or any other kind of `<span>`). May block future extensions.

To be honest, i don't think such simplification will be more easy than "partial implementation of true text node". But if you know concrete problems - let me know. I'll try to find acceptable [cheap] workarounds.

I stay with assumption "light partial implementation is always possible". If my suggestion looks "difficult" - this may be communication problem (i could not express idea good enough). Don't hesitate to ask clarification.

kisvegabor, 2020-08-04:
I got the idea.

Just to be sure: consider the `text` as a child element if exists but if there are "real" children ignore `text`?

puzrin, 2020-08-04:
> Just to be sure: consider the `text` as a child element if exists but if there are "real" children ignore `text`?

That's one of correct possible solution for current moment. If you can force behave it as real single child - no problem. But note, it's still a "temporary kludge", it should not become a base for extents like :before/:after.

I would prefer "true" single text node, if possible (more explicit, more easy to extend in future). But i can not estimate exact cost, and will not push.

puzrin, 2020-08-04:
Memos for future.

### Currently no `display:inline` supported.

It's the main cause of all kinds of hacks with text. And the only possibility to stop hacking - implement partially `display:inline`.

### Mixed content review

`DIV` can keep mixed kinds of elements (both inline and block). No limits to create really shitty things :). Edge cases: 

- Unwrapped text + DIV => human error, don't do that
- Mixed text + buttons => usually human error, don't do that.
- Multiple buttons => organize with flexbox or grid (in html - `li` is preferable for semantics)
- Text + image (smiles inside text and so on) => specific, i would not support
- `float:left`, `float:right`
  - when used for layout (no flexbox support) => use flexbox (or grid) in our case.
  - when used to inject image into multi-paragraph text => no alternative available :(.

If embeddeds can be ok wihout `float` - everything become more simple.

### Possible solution - create container for inline elements only

Let's split cases:

- Container's inner can have only another sub-containers
- Container's inner can have only inline components

If we drop mixed cases - everything become much more simple. That's a first step of simplification.

Is it required to create explicit separate type of container for inlines? May be, but not mandatory (can be hidden from user). For example, `DIV`-s behaviour can be defined by first child (div can ignore all siblings of another type). In this case:

- DIV data stays the same
- It will use different functions to organize inner layout & calculate occupied height

Benefits:

- code for blocks and inlines will be isolated
- possible to extend inlines (add `<span>` support) without risk to fuckup everything.

### Pending

Need to analyse behaviour of all inline nodes

- text
- (later) span-like inline containers (for bold, italic, color, [size?])
- ? button
- ? image

Images & buttons can be supported for "single-child mode" only (when flexbox/grid available). Not beautiful, but may be acceptable.

PS. Worth to check what QT framework does with such cases.

kisvegabor, 2020-08-05:
> Currently no display:inline supported.

We rather need `inline-block`, don't we? AFAIK it's like inline (automatically set size) but allows custom size too.

In our current concept we can simulate inline and block this was:
- inline: set width and height to auto
- block: set width to 100% and height to auto

`auto` means let LVGL to determine the size. Normally it means set according to content size but it also allows for a layout to stretch the size.

Do you think it's significantly different from how display:block/inline/inline-block works?

> Mixed content review

I agree with all.

> Possible solution - create container for inline elements only

Why is it so difficult to support both at the same time? E.g. have a container with a label, a button (inline-like elements) and a container for a list below them. 
The only issue I see is in case of the nested containers the sizes might depend on the size of children containers whose size might depend on the layout. But this issue also persists if a container can have only containers children.

> Need to analyse behaviour of all inline nodes
> text
Clear, to create a text do something like:
```c
lv_obj_t * label = lv_obj_create(lv_scr_act(), NULL);
lv_obj_set_text(obj, "Hello");
```
> (later) span-like inline containers (for bold, italic, color, [size?])

Conceptually seems possible to support later in some way. 

> Button

Visually a button is like a label with a background. It has some additional logic but it's another story. 

Question: how to create a button with icon + text? E.g.
[image]

1. Use symbol in text `lv_obj_set_text(button, LV_SYMBOL_FOO " Open")`
2. Create 2 children
3. Use `before`. Not discussed in detail so far and can be postponed IMO.
4. Allow multiple text nodes. Too complicated now
5. Set text and create an image child. Render text node and children too but always consider text node as the first or last children. 

> Image

What about having `lv_obj_set_image()` function besides `lv_obj_set_text()`? One overwrites the other.  

I think you also meant this by *Images & buttons can be supported for "single-child mode" only*.

embeddedt, 2020-08-05:
> how to create a button with icon + text?

I think the way HTML does it is the most user-friendly: create an image and then put a text node beside it. Unfortunately, this requires support for multiple text nodes, as you noted. I guess a compromise would be to only support that if there's a single text node at the beginning or end. That leaves room to add the functionality in the future.

> What about having `lv_obj_set_image()` function besides `lv_obj_set_text()`? One overwrites the other.

It seems a bit confusing. I'd personally prefer sticking with `lv_img` for that.

puzrin, 2020-08-05:
> We rather need `inline-block`, don't we? AFAIK it's like inline (automatically set size) but allows custom size too.

I would postpone that until multiple inline nodes supported. With single inline node this is not useful

> Why is it so difficult to support both at the same time? E.g. have a container with a label, a button (inline-like elements) and a container for a list below them.
> The only issue I see is in case of [...]

Confusing title, sorry. I mean different - split logic to isolated parts. Containers for blocks and for inlines can be discussed separate.

If it will be convenient to join back on implementation step - to problem.

> Question: how to create a button with icon + text? E.g.

If you mean inline layout, when button width depends on text - that's a problem without multiple inline nodes support. Possible workarounds are:

- Inject icons into font, and type it as part of text.
- Use button as DIV (block container) with fixed width, and organize internals via flexbox.
- Create custom widget "button_with_icon"

That's ugly, but can help sometime. Also, consider to not use "icons+text" with buttons (old-school style).

> What about having `lv_obj_set_image()` function besides `lv_obj_set_text()`? One overwrites the other.

I have feeling, kludge starts spreading :). I would prefer true text node and true image node.
This may require add base "class" before lv_obj, to keep text node light (not sure).

```
lv_base_obj => lv_obj
lv_base_obj => lv_text_node
```

> I think you also meant this by _Images & buttons can be supported for "single-child mode" only_.

Yes.

kisvegabor, 2020-08-06:
Okay, let's drop the "lv_obj as image" idea. :slightly_smiling_face: 

I think it's not required to use a different type for (single) text node. I see how to handle it during rendering and alignments by storing only a text.

> Confusing title, sorry. I mean different - split logic to isolated parts. Containers for blocks and for inlines can be discussed separate.

In HTML there a "basic layout" which puts inline elements next to each other and makes a line break after block elements. We haven't discussed to follow this concept. In my mind it looks like this:
- Place elements freely (convenient and enough in most of the case)
- Use a layout (grid or flexbox)

So there is no strict concept for inline or block. Just blocks have width = 100%, inlines have width = auto, and both have height = auto.

Do you think it's evil? :smile:

puzrin, 2020-08-06:
After thinking a bit about `<button>` with icon + text. "Button" specifics seems only about semantics and properly route clicks when form should be send. In our case (layout/render) it behaves exactly as DIV or SPAN (container, can be tuned via syles).

So, no need to investigate button separate (in pending things), when design behaviour of multiple inline nodes. Need only this things:

- text
- span (inline container)
- image

**One more idea** - can restrict support of multiple inline nodes only to single line (no wrap). Wrapping - only for single text node. This should simplify calculations significantly AND allow cover most of existing needs about icons prepend.

kisvegabor, 2020-08-10:
> can restrict support of multiple inline nodes only to single line (no wrap). Wrapping - only for single text node. This should simplify calculations significantly AND allow cover most of existing needs about icons prepend.

Sounds good. However, if there is a layout (grid or flexbox) it can rearrange these children too right?

puzrin, 2020-08-10:
> Sounds good. However, if there is a layout (grid or flexbox) it can rearrange these children too right?

They operates with containers. I don't understand how this can be related to inline content.

kisvegabor, 2020-08-10:
You have more experience in this filed and probably you are right but e.g. grid arranges `<span>`s (inline elements) and `<div>`s (block elements) in the same way.
See https://codepen.io/kisvegabor/pen/qBZddPx

Am I misunderstanding something?

puzrin, 2020-08-10:
span is NOT guaranteed to be inline. It has `display:inline` by default, but you can style those as block (not good idea). Probably, grid can override span behaviour. I did not used grid and can't say certain.

When i say "inline node" i mean explicit "inline use". If you use those as blocks, then those are not "inline" items.

kisvegabor, 2020-08-10:
Just to clarify: "explicit inline use" means just add some plain text to e.g. a div, right?

If so, see this: https://codepen.io/kisvegabor/pen/qBZddPx
"Plain texts" behave like `<span>`. Flexbox also doesn't make any difference.

puzrin, 2020-08-10:
> Just to clarify: "explicit inline use" means just add some plain text to e.g. a div, right?

Yes. Excluding special case like `<br>`. Span by default is expected to be inline (container for other inline nodes). You can force it to behave different via styles, but that's not related to things i described.

> Flexbox also doesn't make any difference.

When you need next level of madness, see `display:inline-flex` :)

kisvegabor, 2020-08-11:
Okay, thanks! 

A lot of things were considered so let's draw a conclusion. This is how I imagine the first implementation of a text node:
- merge label feature to lv_obj and remove `lv_label`
- add `lv_obj_set_text`, `lv_obj_set_text_static`, `lv_obj_set_text_fmt` functions.
- we can review to other label features (e.g. auto-scrolling, and dots on overflow)
- allow only 1 text node/object, if you need more create "real" children
- the text is always displayed at the 0,0 position (padding is respected) and can't be moved
- layout ignores the text node
- both the text and children are displayed, it's the user's responsibility to choose/use what he needs
- the text has a size, can cause overflow and taken into account with `width: content`-like settings.

puzrin, 2020-08-11:
>  * merge label feature to lv_obj and remove `lv_label`

I think this is right direction, but right now animation can be lost. I don't know how important is that. May be worth mark label as deprecated and remove in next+1 version (but i'm ok to rop immediately).

> * add lv_obj_set_text, lv_obj_set_text_static, lv_obj_set_text_fmt functions.

`lv_obj_set_text_fmt` looks alien for lv_obj. Need to think how to route such demands via custom drawer and so on.

> * allow only 1 text node/object, if you need more create "real" children.

Do i understand right, "true" text node is lv_obj-based. But if DIV has single text node, this can be hacked by assign text content directly to some div property?

> * the text is always displayed at the 0,0 position (padding is respected) and can't be moved

Looks suspicious. Text has "line height" and AFAIK displayed in the middle by default, this can be changed.

> * layout ignores the text node

What do you mean? Text lines height + wrapping may defines parent container height. Even if you disable wrap, this can't be ignored.

> * both the text and children are displayed, it's the user's responsibility to choose/use what he needs

User should not be allowed to use both in the same time, or should be warned somehow. If you give unlimited freedom, after 1-2 years we will have unpleasant demands "from the past", like with new font convertor.

---

One more clarification. When designing architecture, it's important to care about "correct way" first, then apply simplifications/optimizations. In our case:

- First, make sure all proposals are ok with "text node"
- Second - check against "text property".

IMO, we could discuss "text node only".

kisvegabor, 2020-08-12:
> I think this is right direction, but right now animation can be lost. I don't know how important is that. May be worth mark label as deprecated and remove in next+1 version (but i'm ok to drop immediately).

Sure, we can keep it for a while as deprecated.

> lv_obj_set_text_fmt looks alien for lv_obj. Need to think how to route such demands via custom drawer and so on.

Why is it more alien than `lv_obj_set_text`? It just avoids manual `sprintf`-ing.

> Do i understand right, "true" text node is lv_obj-based. But if DIV has single text node, this can be hacked by assign text content directly to some div property?

Yes. 
"light text node" with minimal control:
```html
<div>
text
</div>
```

"true text node" with fulll control:
```html
<div>
<span>text</span>
</div>
```
> Looks suspicious. Text has "line height" and AFAIK displayed in the middle by default, this can be changed.

I meant there is no control to place the text node to arbitrary x/y coordinates on its parent.
Same as in HTML. You can't have `left` on the text of `<div>text</div>` but only on the `div`. 

It just came  to my mind `text-align` can move the text horizontally. It also should work in LVGL too.

So: no explicit control to set arbitrary x/y coordinates.  

> User should not be allowed to use both in the same time, or should be warned somehow. If you give unlimited freedom, after 1-2 years we will have unpleasant demands "from the past", like with new font convertor.

I can really imaginie questions like "why can't I place the next below the chart?" So,  ok, let's ignore text if there are real children.

> What do you mean? Text lines height + wrapping may define parent container height. Even if you disable wrap, this can't be ignored.

I meant if - let's say - grid is applied on a container it won't place the text node to cells. It arranges only real children.  The text node still has size and the parent sees this size to show scrollbars or set its height accordingly. This point is only about explicit layouts.

It is resolved if you can't mix text node with real children.

puzrin, 2020-08-12:
> > lv_obj_set_text_fmt looks alien for lv_obj. Need to think how to route such demands via custom drawer and so on.
> 
> Why is it more alien than `lv_obj_set_text`? It just avoids manual `sprintf`-ing.

Reason is the same why it doesn't cook coffee. Because we should add everything possible. May be i misunderstood what is `lv_obj_set_text` about and it should be removed too.

Note, i'm not against formatter helper. But i'm against pinning it hard into node. Need to find more flexible way to use "extenal string builer-s". For example, i use ETL for strings & formatting.

> I meant there is no control to place the text node to arbitrary x/y coordinates on its parent.
> Same as in HTML. You can't have `left` on the text of `<div>text</div>` but only on the `div`.
>
> It just [...]

Thanks, now  understand. I think, advanced horizontal positioning is not needed right now. Later, if you wish horizontal scroll animation, you may wish to add `text-indent` https://stackoverflow.com/questions/40789222/css-animate-scrolling-text-using-text-indent.

Vertical align is not critical now too. At least, until icons-before-text used. Can't say certain about future.

So: starting with the most simple mode is ok, result will be useable for most cases.

> I meant if - let's say - grid is applied on a container it won't place the text node to cells. It arranges only real children.  The text node still has size and the parent sees this size to show [...]

Technically, if you mix "text + div", text will behave like it has virtual own div wrapper. And layout works with "div(text) + div".

- I can't comment about lvgl... but in webev only unlearned beginners can do such thing (unwrapped text + div), and that will be rejected on nearest code review :)
- Implementing such thing would be waste of time and promoting bad practice IMO. I think, if docs explain supported/rejected features, user should understand.

I vote to prohibit "text + div" sibblings. Let's stay with:

- All nodes can be of block-like or inline-like type.
- Only nodes of the same kind allowed in sibblings. Not matched nodes are ignored.

kisvegabor, 2020-08-12:
> Note, i'm not against formatter helper. But i'm against pinning it hard into node. Need to find more flexible way to use "extenal string builer-s". For example, i use ETL for strings & formatting.

No problem with that. E.g. 
```c
char buf[100];
my_printer(buf, <your code>);
lv_obj_set_text(obj, buf); //Any text from a buffer
lv_obj_set_text_static(obj, const_string); //Save pointer only
lv_obj_set_text_fmt(obj, "num: %d", x); //printf-like syntax
```

> Thanks, now understand. I think, advanced horizontal positioning is not needed right now.

Not that advanced just left/right/center. That's very minimal IMO. Without it, you can't center multiline texts.

>  Later, if you wish horizontal scroll animation, you may wish to add text-indent https://stackoverflow.com/questions/40789222/css-animate-scrolling-text-using-text-indent.

See the list element with the long label on Selectors tab here: https://lvgl.io/demos
The beginning of the text is shown after the end of the text. 

> Vertical align is not critical now too. At least, until icons-before-text used. Can't say certain about future.

I agree to skip it for now. You can create real children if it's needed.

> Technically, if you mix "text + div", text will behave like it has virtual own div wrapper. And layout works with "div(text) + div".

Okay, we are on the same page. :) 

> All nodes can be of block-like or inline-like type.
> Only nodes of the same kind allowed in sibblings. Not matched nodes are ignored.

Here I checked the list of inline elements: https://developer.mozilla.org/en-US/docs/Web/HTML/Inline_elements

We don't have all this stuff but only 2 things:
- implicit text node without siblings
- everything else is a block. E.g. all these are blocks:
```html
<div>
  <div>text</div>
  <button>text</button>
  <table>...</table>
</div>
```

`<img>` is an exception but for simplicity it can be considered as block in our case.

puzrin, 2020-08-12:
> ```c
> char buf[100];
> my_printer(buf, <your code>);
> lv_obj_set_text(obj, buf); //Any text from a buffer
> lv_obj_set_text_static(obj, const_string); //Save pointer only
> lv_obj_set_text_fmt(obj, "num: %d", x); //printf-like syntax
> ```

User can "manually" sprintf to `buf` anytime when update desired. Seems `lv_obj_set_text_fmt(obj, "num: %d", x)` not needed (only expose sprintf-like helper in utils if required).

`lv_obj_set_text` & `lv_obj_set_text_static` - we always store static pointer to string. BUT content by pointer may change, if target is in ram. Are both methods still needed? Is that C specific? I verify only data/flow and can miss something about coding.

> See the list element with the long label on Selectors tab here: [lvgl.io/demos](https://lvgl.io/demos)
> The beginning of the text is shown after the end of the text.

I see 3 methods of resolution (not urgent):

- Keep custom widget (~ like now, inherit text & add animation effects)
- Replace effect, do `<= =>` animation instead of circular.
- Say user "use 2 texts or custom drawer" to emulate circular move.

If "lv_label" not removed immediately - let's postpone and decide later.

> Here I checked the list of inline elements: [developer.mozilla.org/en-US/docs/Web/HTML/Inline_elements](https://developer.mozilla.org/en-US/docs/Web/HTML/Inline_elements)
> 
> We don't have all this stuff but only 2 things:
> 
> * implicit text node without siblings
> 
> * everything else is a block. E.g. all these are blocks:
> 
> 
> ```
> <div>
>   <div>text</div>
>   <button>text</button>
>   <table>...</table>
> </div>
> ```
> 
> `<img>` is an exception but for simplicity it can be considered as block in our case.

I don't agree. Let's clarify what we need in ideal world:

- text
- `<button>` should be inline by default, like `<span>` (user may wish minimal width [defined by text]).
- `<img>` should be inline (feature icon-before-text)
- support of multiple inline sibblings without line wrap (and single text with line wrap)

If you wish postpone/drop multiple inlines, change of button/img to block-like still may cause too many side effects. May be, only with big red warning "feature incomplete and will be changed in next+1". But why to release in such state then? User can take legacy widgets for block-like behaviour.

Please, clarify your intent. Suggestion about img/button change looks suspicious and fragile. May be this may work somehow, but need to verify twice.

kisvegabor, 2020-08-13:
> User can "manually" sprintf to buf anytime when update desired. Seems lv_obj_set_text_fmt(obj, "num: %d", x) not needed (only expose sprintf-like helper in utils if required).

It's more than manul `sprintf`-ing. With `sprintf`
- you need to know to the length of the result string to allocated a buffer. (`char buf[???]`)
- you need memory for the temporal buffer of sprintf. 

`lv_label_set_text_fmt` calculates the required length first, dynamically allocates the buffer, and "computes" the string there.

> lv_obj_set_text & lv_obj_set_text_static - we always store static pointer to string.

That's not true. E.g. a text area always stores a dynamic text. `lv_obj_set_text` allocates RAM for the text, while `lv_obj_set_text_static` saves only a pointer.

>> See the list element with the long label on Selectors tab here: lvgl.io/demos
The beginning of the text is shown after the end of the text.

> Replace effect, do <= => animation instead of circular.

If it works the circular is 3 lines of code to draw the label again with an offset. Now, this scrolling is one of the several "long modes" (not the best name): crop: expand, dots, brake (should be wrap), scroll, scroll circularly. We can consider what are required here. 
Some though inspired by CSS, but it can really be postponed:
- `text-overflow: clip | ellipsis-start | ellipsis-end | ellipsis-middle | scroll | scroll-circle`
- `word-wrap: true | false`: 
   - if width is fix and false -> apply `text-overflow`
   - if width is fix and true -> multiline
      - if height is fix -> apply text-overflow
       - if height is `auto/content` -> expand
   - if width is `auto/content` -> expand

> I don't agree. Let's clarify what we need in ideal world:

Honestly, I don't think we need the concept of `inline` at all.  Af far as I understand its purpose is to automatically put elements in a row. However, with a powerful and easy to use layout system (grid) it can be easily mimicked with much more control. E.g. for the button with icon and text: 
```c
lv_obj_set_grid_template_column(btn, "auto auto");  //Just example, not API proposal :)
lv_img_create(btn);
lv_label_create(btn);
```
1 line of code and not bother with the `inline`/`block` question because everything is `block`.

What do you think?

puzrin, 2020-08-13:
> It's more than manul `sprintf`-ing. With `sprintf`
>
>  * you need to know to the length of the result string to allocated a buffer. (`char buf[???]`)
>  * you need memory for the temporal buffer of [...]

1. User MUST know buffer size anyway, to cache result. In most cases you can declare `static buf[20] = ''` inside function, having to return formatted string. That's not a problem.
2. If `lv_obj_set_text` already cares about clone & memory allocaion, this can't be declared as avantage of `lv_obj_set_text_fmt`. User can just call sprinf to local buf + `lv_obj_set_text`.

> Honestly, I don't think we need the concept of `inline` at all. Af far as I understand its purpose is to automatically put elements in a row.

IMO it has 2 purposes:

1. Set parent width by content (when no wrap)
2. In our case - make implementation more modular.

> However, with a powerful and easy to use layout system (grid) it can be easily mimicked with much more control.

div / grid / flex DON'T USE content to calculate width. You will have to implement `inline-flex` or `inline-grid`. That's significantly more complicated than inline (or inline-block).

IMO you tend to search "magical recipe which solves all problems at once". I'd prefer something more common and predicable in v8.

kisvegabor, 2020-08-13:
**lv_obj_set_text_fmt**
With plain sprintf you would need this:
```c
size_t len = sprintf(NULL, "%d:%d", hour, min);   //Jus to get the final size

//If compiler supports VLA (Visual Studio doesn't)
char buf[len + 1];                 //Don't forget the +1
//If VLA is not supported by the compiler
char * buf = malloc(len + 1);      //Don't forget the +1

sprintf(buf, "%d:%d", hour, min);   //Repeat the expression but now use a buffer
lv_label_set_text(label, buf); 
 //At this point both `buf` and label->text consumes memory.

free(buf);   //If it was allocated
```
So it's counterintuitive, error-prone, and uses double memory. I think, for cases like this it's fine to have a dedicated function.

---

**Inline**
> Set parent width by content (when no wrap)

I think we can do it with `width:content` as well.

> In our case - make implementation more modular.

Introducing another element placing method makes implementation more fragile to me.

> IMO you tend to search "magical recipe which solves all problems at once".

Yes, that's true. :slightly_smiling_face:  It'd be the best if all of our layout problems could be solved by one clean feature. No problem is it's complex or complicated to implement if it works well, versatile, scales well (from simple to complex layouts) and easy to use.

My problem with bringing inline into the game is that they have very limited use cases (at least what I see). Place content sized elements after each other. It's just too specific. If you can mention some real-life cases where inline fits very well and grid or inline-grid isn't, I'd be very curious to see them.

puzrin, 2020-08-14:
This just works:

```c
void format_hm(uint8_t h, uint8_t m) {
  char buf[20];
  sprintf(buf, "%d:%d", h, m);
  lv_label_set_text(label, buf); 
}
```

cached:

```c
void format_hm_cached(uint8_t h, uint8_t m) {
  static char buf[20];
  static uint8_t prev_h = 255, prev_m;

  if (prev_h != h || prev_m != m) sprintf(buf, "%d:%d", h, m);

  lv_label_set_text_static(label, buf); 
}
```

With ETL you can get the same but with bounds check(safe) https://www.etlcpp.com/to_string.html. Or `sprintf_s` in C11.

> My problem with bringing inline into the game is that they have very limited use cases (at least what I see). Place content sized elements after each other.

Syntax color, partial bold/italic, insert icon, ellipsis, overflow.

Final nodes with "content" (not decoration) are ALWAYS inline. If you reduce features to primitive - they still stays inline (and can be extended later).

>  If you can mention some real-life cases where inline fits very well and grid or inline-grid isn't, I'd be very curious to see them.

```html
<span class="icon_home">icon_code</span>Go home
```

If i wish to insert icon i just wrap it and assign class (to adjust offsets). With inline-grid i will have to wrap each sibbling & assign grid to parent. That should work, but more efforts & un-natural.

You wish to declare inlines implicit, as allowed in inline-grid only (that's NOT the same as grid, but separate additional feature). But saying then "inlines not needed" adds huge confusion in communication.

- Explicit declaration is usually [always :)] better that implicit.
- If you are able to implement `inline-grid` right, it would be strange to cut smaller subset as inline/inline-block.

In short: grid + inline-grid can be used everywhere, but this approach is not for humans :)

kisvegabor, 2020-08-17:
> void format_hm(uint8_t h, uint8_t m) {

You are still using extra memory for `buf` and you need to know the resulting string's length in advance.

> Syntax color, partial bold/italic,

These are based on the same feature IMO. multiple "text parts" (`<span>`) in a larger text. We have postponed it. 

>  insert icon

It can be built from an image an text if we are not supporting multiple text nodes now. 

> ellipsis, overflow.

I can't see how they are related.  It still produces overflow or ellipsis even it's a block:
```html
<div class="width-50px">
blah blah blah blah blah blah blah blah 
<div>
```

The only real use case I see is placing texts/images next to each other. However there so many alignment settings here that are typically complicated in CSS. E.g. horizontal centering + vertical center. With a grid, you can do all these with ease. Why develop and maintain a more limited still complicated new alignment rules besides a really flexible one? I think the only question is to find an easy to use API for the grid. Some predefined layouts from macros also can help to solve everyday tasks.
 
Form our needs, as I see now, `inline-grid` is like `grid` but it supports `width/height:auto`.

puzrin, 2020-08-17:
> You are still using extra memory for `buf` and you need to know the resulting string's length in advance.

1. Memory on stack is auto-managed, and should not be considered as problem.
2. In each concrete place you have length estimate. It's no manatory to know exact length,  good estimate (with reserve) and overflow protection can be ok.

> These are based on the same feature IMO. multiple "text parts" (`<span>`) in a larger text. We have postponed it.

Yes, we postponed, but you asked examples "why inlines needed".

> With a grid, you can do all these with ease. Why develop and maintain a more limited still complicated new alignment rules besides a really flexible one?

1. Because in simple cases it requires more efforts to define properties.
2. Because only you believe in possibility to do grid right in v8 :). But i would prefer more guaranteed result.

> Form our needs, as I see now, `inline-grid` is like `grid` but it supports `width/height:auto`.

Similar can be said about any `inline-*`. But simple description != simple implementation, for this case.

embeddedt, 2020-08-17:
> Memory on stack is auto-managed, and should not be considered as problem.

I think it's better to avoid the stack if possible, as not everyone has memory to devote to an 8K stack. Thus far, we've tried to design the APIs in a way that limits stack usage. There's already a lot of recursion for rendering.

puzrin, 2020-08-17:
> I think it's better to avoid the stack if possible, as not everyone has memory to devote to an 8K stack. Thus far, we've tried to design the APIs in a way that limits stack usage. There's already a lot of recursion for rendering.

I suggest to leave implementation to user. I also provided example with `static` buffer (for cache).

Point of discussion was a bit different - in my eyes formatter looks alien to text node built-in features. I would re-consider it to add only after everything else works.

kisvegabor, 2020-08-17:
**Text format**
As it solves a non-trivial problem, I'd like to keep this function. 

**Inline** 
I don't know all the details of grid issues and don't see exactly how the API should look like either. I have some projects to work on and couldn't dive into these questions yet. 

I really like to go with one powerful, flexible, and easy to use layout system. Right now I think grid can work. My suggestion is to implement grid and if something is difficult to solve with it we can start to think about the inline part. As grid overwrites the default inline layout, inline can be discussed separately. 

@embeddedt What do you think about this inline/grid topic? Do you also think we need inline mechanism in LVGL?

puzrin, 2020-08-17:
>  As it solves a non-trivial problem, I'd like to keep this function.

I'm not against this function, i'm against putting it into node. The most complicated part is `sprintf_s()` implementation with optional float. But this can stay in utils. The rest can be postponed, until we invent something better or stabilize essential methods.

> I really like to go with one powerful, flexible, and easy to use layout system...

20 years ago i also believed in such things :). Now, i don' believe in "big monolytic solutions, which should solve all problems a once". And i'm a bit sceptical about doing such things wihout tests.

Anyway, i'm not going to demotivate you or say this is impossible. This is possible. But i speak about deadlines. I'm sure this could be ok in v10. May be in v9. But not sure about v8. I hope, you believe in my good intent. All i did in this thread - carefully reduced subset of CSS to be possible implement via smaller slices. If that helps - ok, if not - no problem.

embeddedt, 2020-08-18:
> What do you think about this inline/grid topic? Do you also think we need inline mechanism in LVGL?

I'm much more familiar with flexbox than grid, as the projects I've worked on so far require IE11 compatibility and it doesn't support grid. However, I will look into grid some more so that I can have a more educated opinion. :wink:

If everything that's possible to do in flexbox is possible with grid (easily), then I don't think we necessarily *need* inline support.

Bold/italic and icon nodes can be [imitated](https://codepen.io/embeddedt/pen/poybqwR) using flexbox. The syntax is not that developer-friendly, but it can be done.

I think it really boils down to how much flexibility is needed in an embedded GUI. I highly doubt anyone needs anything more complex than [this type of layout for text](https://codepen.io/embeddedt/pen/qBZNLpM). That means that the only use for inline support is multiple text nodes.

As long as we architect the API in such a way that inline support can be added later, I don't see any issue with developing grid/flexbox first, as long as our current layout system with `lv_cont` continues functioning.

kisvegabor, 2020-08-18:
@puzrin 
> I'm not against this function, i'm against putting it into node.

Finally, we won't have `lv_label`, only the node. So we have no other place to place this function.

> The most complicated part is sprintf_s() implementation with optional float. 

Float support can be disabled in [lv_conf.h](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L519)

> And i'm a bit sceptical about doing such things wihout tests.

We can have tests for it. :slightly_smiling_face: 

> I hope, you believe in my good intent.

I have no doubt about it and I'm really happy that you take the effort to discuss these things. Thank you very much! :slightly_smiling_face: 
August will be short for me because of holidays, family stuff, etc. I hope in September I can dive into it. We will see how it goes. If it will look really so time-consuming and complicated then we can reconsider its schedual. 
Now I feel excited about having such a great thing like a grid layout.  In my currently running projects, I see its usefulness all the time. So I can't stop now. :laughing:

> If everything that's possible to do in flexbox is possible with grid (easily), then I don't think we necessarily need inline support.

I saw only missing thing from grid compared to the flexbox. The grid needs to look like a grid. So it doesn't work automatically (you can still explicitly place items on a 12 column grid):
```
short text | much much longer text
x | a really much much longer text
```

> I think it really boils down to how much flexibility is needed in an embedded GUI. I highly doubt anyone needs anything more complex than this type of layout for text. That means that the only use for inline support is multiple text nodes.

I agree.

> architect the API in such a way that inline support can be added later, I don't see any issue with developing grid/flexbox first, as long as our current layout system with lv_cont continues functioning.

Finally, I think `lv_page`, `lv_cont` and `lv_label` can be removed in v9. Do you agree?

puzrin, 2020-08-19:
> Bold/italic and icon nodes can be [imitated](https://codepen.io/embeddedt/pen/poybqwR) using flexbox. The syntax is not that developer-friendly, but it can be done.

No sure in correctness of example:

1. If you replace `button` with `div` - result will be different
2. Use of `span` is invalid, should be `div` to, if yo speak about emulation

In other words, this example does no reflect situation with missed inlines and can not prove anything.

You can NOT imitate inline behavior via `flex`. You will need `inline-flex` (in real world `inline-block` without line wrap will be ok).

> Finally, I think `lv_page`, `lv_cont` and `lv_label` can be removed in v9. Do you agree?

This should happen sometime. May be some users will cry about missed animation of long labels.

kisvegabor, 2020-08-20:
> This should happen sometime. May be some users will cry about missed animation of long labels.

IMO text node should support it :slightly_smiling_face:

kisvegabor, 2020-08-24:
I'd say let's focus on the other questions that came up regarding the architectural rework and we will see how to proceed with the text node question.

upbeat27, 2020-09-04:
Please keep animation of long labels (scrolling).
LVGL isn't a web browser. 
It is likely to be used on small screens. 
Text often cannot all fit on a small screen.
Please think "is this useful?" rather than "is this how HTML/CSS works?" before throwing features away.
Just my 2 cents.

kisvegabor, 2020-09-28:
I added it as an idea to the [ROADMAP](https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md).

Let's warn up this topic a little bit later.

closed 2020-09-28 by kisvegabor